horizon-eda/horizon #3: ./horizon-pool-update seems to leak memory

rnestler, 2017-01-30:
When I try to run ./horizon-pool-update it just uses more and more RAM (I killed it when it used 10GB...)

carrotIndustries, 2017-01-30:
Looks very much like an issue related to walking the pool directories.
Can you run it in gdb and provide backtrace at some random point when it gets stuck?

rnestler, 2017-01-30:
```
#0  0x0000000000415622 in nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator>::lexer::scan() (this=0x7fffffffb320) at ./json.hpp:7967
#1  0x0000000000411922 in nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator>::parser::get_token() (this=0x7fffffffb2f0) at ./json.hpp:9232
#2  0x000000000040ed97 in nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator>::parser::parse() (this=0x7fffffffb2f0) at ./json.hpp:9022
#3  0x000000000040e40a in nlohmann::operator>>(std::istream&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator>&) (i=..., j=...) at ./json.hpp:6139
#4  0x0000000000437b4a in horizon::Part::new_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, horizon::Pool&) (filename="/home/roughl/proggen/projects/horizon-pool/parts/", pool=...) at pool/part.cpp:107
#5  0x00000000004d5948 in horizon::Pool::get_part(horizon::UUID const&) (this=0x7fffffffd4e0, uu=...)
    at pool/pool.cpp:78
#6  0x0000000000436b2d in horizon::Part::Part(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator> const&, horizon::Pool&) (this=0x7fffffffc2e0, uu=..., j=..., pool=...)
    at pool/part.cpp:14
#7  0x0000000000437bb3 in horizon::Part::new_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, horizon::Pool&) (filename="/home/roughl/proggen/projects/horizon-pool/parts/passive/resistors/cpl/CPL-RES-0603-3.6K-0.1W.json", pool=...) at pool/part.cpp:109
#8  0x00000000004edd13 in update_parts(SQLite::Database&, horizon::Pool&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (db=..., pool=..., directory="/home/roughl/proggen/projects/horizon-pool/parts/passive/resistors/cpl", prefix="passive/resistors/cpl") at pool-update/pool-update.cpp:199
#9  0x00000000004edfce in update_parts(SQLite::Database&, horizon::Pool&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (db=..., pool=..., directory="/home/roughl/proggen/projects/horizon-pool/parts/passive/resistors", prefix="passive/resistors") at pool-update/pool-update.cpp:218
#10 0x00000000004edfce in update_parts(SQLite::Database&, horizon::Pool&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (db=..., pool=..., directory="/home/roughl/proggen/projects/horizon-pool/parts/passive", prefix="passive") at pool-update/pool-update.cpp:218
#11 0x00000000004edfce in update_parts(SQLite::Database&, horizon::Pool&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (db=..., pool=..., directory="/home/roughl/proggen/projects/horizon-pool/parts", prefix="") at pool-update/pool-update.cpp:218
#12 0x00000000004eeb81 in main(int, char**) (c_argc=1, c_argv=0x7fffffffdb38)
    at pool-update/pool-update.cpp:274
```

carrotIndustries, 2017-01-30:
Oh, I see
````
/home/roughl/proggen/projects/horizon-pool/parts/
````

Try the latest commit: https://github.com/carrotIndustries/horizon/commit/3b199795174a452e9ef02b3c00a38804771da19f  At least, it should blow up with an exception.

Looks like I really need to rethink how the parts get updated.

carrotIndustries, 2017-11-07:
Closing due to old age, probably fixed by now.

closed 2017-11-07 by carrotIndustries