horizon-eda/horizon #8: Create packages for some linux distros.

binaryplease, 2017-06-23:
Hi, 

I would like to try this out on linux arch, but it would be nice to have a package or at least a AUR package.
Would you be interested to create one?

carrotIndustries, 2017-06-23:
I'll probably come up with a PKGBUILD when horizon is reasonably stable. Right now, horizon may frequently change in incompatible ways (see past commits), so having a ready-to-install package may raise false expectations.

binaryplease, 2017-06-23:
Ok, thanks for the info, I'll try to install by hand then later.

janhieber, 2017-10-25:
Made a PKGBUILD.
I did not upload it to AUR yet, if someone wants to maintain it, feel free to do so.
Also 'i686' and other archs should be added after testing them.

```bash
_name=horizon
pkgname=${_name}-git
pkgver=r185.59b2f6b
pkgrel=1
epoch=
pkgdesc="free EDA package written in C++ with GTK3"
arch=('x86_64')
url="https://github.com/carrotIndustries/horizon"
license=('GPL')
groups=()
depends=('boost-libs' 'boost' 'glm' 'yaml-cpp' 'zeromq' 'gtkmm3' 'cairomm' 'librsvg' 'sqlite')
checkdepends=()
optdepends=()
provides=()
conflicts=()
replaces=()
backup=()
options=()
install=
changelog=
source=("git+https://github.com/carrotIndustries/horizon.git")
noextract=()
md5sums=('SKIP')
validpgpkeys=()

pkgver() {
    cd "$srcdir/$_name"
    printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

build() {
    cd "$srcdir/$_name"
    make
}

package() {
    for f in horizon-imp horizon-pgm-test horizon-pool horizon-pool-mgr horizon-pool-update-parametric horizon-prj horizon-prj-mgr
    do
        install -Dm755 $srcdir/$_name/$f $pkgdir/usr/bin/$f
    done
}
```

carrotIndustries, 2017-10-25:
Thanks for the template. I adjusted a few things and uploaded it: https://aur.archlinux.org/packages/horizon-git/ But beware, although horizon is way more polished and usable than it was some months ago, things are still changing.

closed 2017-10-25 by carrotIndustries