horizon-eda/horizon #9: Windows exe display issue

boyddotee, 2017-10-25:
Interesting project, has a few issues with display on my system the top menu buttons appear behind the windows application bar.

[image]

Possibly relevant info: the computers graphics card is a GTX 1050 with the resolution at 1910x1080 i also have the the windows recommended scaling disabled.

carrotIndustries, 2017-10-25:
Someone has reported a similar issue to me in person: He was using a laptop with nvidia and intel gpu. When running the application using the intel gpu, it looked just like your screenshot. With the nvidia one, it was fine.

>i also have the the windows recommended scaling disabled.

So did you set the scaling to 100% in settings? Have you tried the various DPI scaling option in the compatibility tab for horizon-imp.exe? 

Does this issue also appear with the pool manager? This one uses the OpenGL canvas as well, but uses client side decorations. To me, this looks like some code within Gtk gets the window size wrong.

boyddotee, 2017-10-26:
> Someone has reported a similar issue to me in person: He was using a laptop with nvidia and intel gpu. When running the application using the intel gpu, it looked just like your screenshot. With the nvidia one, it was fine.

Turns out it was running on the intel chipset (HD 630 thought i had disabled it, its also a laptop), with the effect not occurring when ran on the nvidia gtx.

> So did you set the scaling to 100% in settings? Have you tried the various DPI scaling option in the compatibility tab for horizon-imp.exe?

Thats the one,  ive just played with the DPI scaling options under the application compatibility with no change. 

> Does this issue also appear with the pool manager?

The pool manager seems to be fine when ran on both;

[image]

carrotIndustries, 2017-10-26:
Oh, I forgot to mention: In the pool manager, open the pool.json and go to the "Packages" tab, so it actually displays the OpenGL canvas.

boyddotee, 2017-10-26:
Seems to work fine on both GPU's image below for reference;

[image]

carrotIndustries, 2017-10-26:
Ok, this confirms my guess that something in the stack tells Gtk the size of the whole window including non client area, so not a DPI scaling issue. This works fine for the pool manager since Gtk draws everything including the decorations.

Not sure whether this is a Gtk or intel driver bug...

carrotIndustries, 2017-11-07:
Since the imp now uses client side decorations this issue is solved.

closed 2017-11-07 by carrotIndustries