horizon-eda/horizon #10: Current git crashes when updating pool

steinm, 2017-11-06:
I just called the pool mgr and updated it. I'm running horizon on debian sid

./horizon-pool-mgr 
SELECT parts.uuid, parts.MPN, parts.manufacturer, packages.name, GROUP_CONCAT(tags.tag, ' '), parts.filename FROM parts LEFT JOIN tags ON tags.uuid = parts.uuid LEFT JOIN packages ON packages.uuid = parts.package WHERE parts.MPN LIKE ? AND parts.manufacturer LIKE ? AND (parts.entity=? or ?) GROUP BY parts.uuid  ORDER BY parts.MPN ASC
SELECT parts.uuid, parts.MPN, parts.manufacturer, packages.name, GROUP_CONCAT(tags.tag, ' '), parts.filename FROM parts LEFT JOIN tags ON tags.uuid = parts.uuid LEFT JOIN packages ON packages.uuid = parts.package WHERE parts.MPN LIKE ? AND parts.manufacturer LIKE ? AND (parts.entity=? or ?) GROUP BY parts.uuid  ORDER BY parts.MPN ASC
hello from thread
tags
entities
symbols
terminate called after throwing an instance of 'std::out_of_range'
  what():  key 'unit' not found
Abgebrochen

carrotIndustries, 2017-11-06:
Did you add any new symbols to the pool? It'd be great if you could provide a backtrace:
    
     gdb ./horizon-pool-mgr
     catch throw
     run
     #when it crashes
     bt

steinm, 2017-11-06:
I suppose I found the reason. I created an entity below the symbols folder instead of the entities folder. I just wonder why the file dialog opened the symbols folder in the first place when saving the entity.

carrotIndustries, 2017-11-06:
That was just a copy/paste issue, but in the long term the pool manager should prevent creating things in the wrong directory.

commit 46023100 (2017-11-06): set correct initial folder in pool manager file chooser

should fix #10
closed 2017-11-06 by carrotIndustries