horizon-eda/horizon #11: Value in parameters seems to have no effect

steinm, 2017-11-06:
When creating a padstack and using parameters, they will not be used under certain circumstances when you apply the script. This happens for example, when you already have a script and parameters but haven't assigned a parameter class to a shape. If you set the parameter class afterwards an apply the script, then it will not change the shape. You have to explicitly change a parameter value and apply the script again.

carrotIndustries, 2017-11-06:
Unfortunately, I can't reproduce the issue as you described it. What are the exact steps you did to cause this behavior?

steinm, 2017-11-06:
Now I'm also unable to reproduce it. Just close this ticket. I'll open a new one, if I run into this again.

closed 2017-11-06 by carrotIndustries