horizon-eda/horizon #16: add install and dist target to makefile

steinm, 2017-11-09:
I propose to add an 'install' and  'dist' target to the makefile. That would ease the creation of debian packages.

PREFIX=/usr
VERSION=$(shell git log -1 --date="format:%Y%m%d" --pretty="format:%cd")

install: install-bin

install-bin:
  install -d $(DESTDIR)$(PREFIX)/bin
  install -m 755 horizon-imp horizon-pool horizon-prj horizon-pool-update-parametric horizon-prj-mgr horizon-pgm-test horizon-pool-mgr $(DESTDIR)$(PREFIX)/bin

dist:
  tar -C .. -cJv --exclude=debian --exclude=horizon-$(VERSION).tar.xz --exclude-vcs --exclude-vcs-ignore --transform='s/horizon/horizon-$(VERSION)/' -f ../horizon-$(VERSION).tar.xz horizon

carrotIndustries, 2017-11-09:
Seems like the dist target assumes the directory the source code lives in is called `horizon`, but looks good otherwise.

carrotIndustries, 2018-02-06:
Obsolete due to upcoming move to cmake

closed 2018-02-06 by carrotIndustries