horizon-eda/horizon #21: Crash when trying to add-part

rnestler, 2017-11-28:
I got a crash when I tried to add a part on a schematics:
```
% gdb ./horizon-prj-mgr 
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./horizon-prj-mgr...done.
(gdb) run
Starting program: ~/projects/horizon/horizon-prj-mgr 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffea33a700 (LWP 16497)]
[New Thread 0x7fffe9b39700 (LWP 16498)]
[New Thread 0x7fffe9126700 (LWP 16499)]
[New Thread 0x7fffe8925700 (LWP 16500)]
[New Thread 0x7fffdbde2700 (LWP 16502)]
[New Thread 0x7fffdb3db700 (LWP 16503)]
[New Thread 0x7fffdabda700 (LWP 16504)]
[New Thread 0x7fffda1cc700 (LWP 16505)]
[New Thread 0x7fffd99cb700 (LWP 16506)]
[New Thread 0x7fffd91ca700 (LWP 16507)]
[New Thread 0x7fffc0ece700 (LWP 16509)]
[Thread 0x7fffdabda700 (LWP 16504) exited]
[Thread 0x7fffdbde2700 (LWP 16502) exited]
[Thread 0x7fffd91ca700 (LWP 16507) exited]
[Thread 0x7fffdb3db700 (LWP 16503) exited]
[New Thread 0x7fffdb3db700 (LWP 16511)]
[New Thread 0x7fffd91ca700 (LWP 16512)]
[New Thread 0x7fffdbde2700 (LWP 16513)]
[New Thread 0x7fffdabda700 (LWP 16514)]
[New Thread 0x7fffa7fff700 (LWP 16515)]
[New Thread 0x7fff9f7fe700 (LWP 16516)]
[New Thread 0x7fffa77fe700 (LWP 16517)]
[New Thread 0x7fffa6ffd700 (LWP 16518)]
[New Thread 0x7fffa67fc700 (LWP 16519)]
[Thread 0x7fffdbde2700 (LWP 16513) exited]
[Thread 0x7fffd91ca700 (LWP 16512) exited]
[Thread 0x7fffdb3db700 (LWP 16511) exited]
[Thread 0x7fffa7fff700 (LWP 16515) exited]
[Thread 0x7fffdabda700 (LWP 16514) exited]
[Thread 0x7fff9f7fe700 (LWP 16516) exited]
[Thread 0x7fffa67fc700 (LWP 16519) exited]
[Thread 0x7fffd99cb700 (LWP 16506) exited]
[Thread 0x7fffa6ffd700 (LWP 16518) exited]
[New Thread 0x7fffa6ffd700 (LWP 16520)]
[New Thread 0x7fffd99cb700 (LWP 16521)]
[New Thread 0x7fffa67fc700 (LWP 16522)]
[New Thread 0x7fff9f7fe700 (LWP 16523)]
[New Thread 0x7fffdbde2700 (LWP 16524)]
[New Thread 0x7fffdb3db700 (LWP 16525)]
[New Thread 0x7fffdabda700 (LWP 16526)]
[Thread 0x7fffdb3db700 (LWP 16525) exited]
[Thread 0x7fff9f7fe700 (LWP 16523) exited]
[Thread 0x7fffa77fe700 (LWP 16517) exited]
[Thread 0x7fffdabda700 (LWP 16526) exited]
[Thread 0x7fffa67fc700 (LWP 16522) exited]
[Thread 0x7fffa6ffd700 (LWP 16520) exited]
[Thread 0x7fffd99cb700 (LWP 16521) exited]
[New Thread 0x7fffd99cb700 (LWP 16529)]
[New Thread 0x7fffa6ffd700 (LWP 16530)]
[New Thread 0x7fffa67fc700 (LWP 16531)]
[New Thread 0x7fffdabda700 (LWP 16532)]
[New Thread 0x7fffdb3db700 (LWP 16533)]
[New Thread 0x7fffd91ca700 (LWP 16534)]
[New Thread 0x7fffa7fff700 (LWP 16535)]
[New Thread 0x7fffa77fe700 (LWP 16536)]
[New Thread 0x7fffa5ffb700 (LWP 16537)]
[Thread 0x7fffdb3db700 (LWP 16533) exited]
[Thread 0x7fffa67fc700 (LWP 16531) exited]
[Thread 0x7fffd99cb700 (LWP 16529) exited]
[Thread 0x7fffdbde2700 (LWP 16524) exited]
[Thread 0x7fffa77fe700 (LWP 16536) exited]
[Thread 0x7fffa5ffb700 (LWP 16537) exited]
[Thread 0x7fffa7fff700 (LWP 16535) exited]
[Thread 0x7fffd91ca700 (LWP 16534) exited]
[Thread 0x7fffa6ffd700 (LWP 16530) exited]
[New Thread 0x7fffa6ffd700 (LWP 16538)]
[Thread 0x7fffdabda700 (LWP 16532) exited]
[New Thread 0x7fffdabda700 (LWP 16539)]
[Thread 0x7fffa6ffd700 (LWP 16538) exited]
SELECT parts.uuid, parts.MPN, parts.manufacturer, packages.name, GROUP_CONCAT(tags.tag, ' '), parts.filename FROM parts LEFT JOIN tags ON tags.uuid = parts.uuid LEFT JOIN packages ON packages.uuid = parts.package WHERE parts.MPN LIKE ? AND parts.manufacturer LIKE ? AND (parts.entity=? or ?) GROUP BY parts.uuid  ORDER BY parts.MPN COLLATE naturalCompare ASC
col 2
create proc
spawn /home/roughl/proggen/projects/horizon/horizon-imp -c /home/roughl/proggen/projects/horizon/projects/foobar/top_sch.json /home/roughl/proggen/projects/horizon/projects/foobar/top_block.json 
[Thread 0x7fffdabda700 (LWP 16539) exited]
[New Thread 0x7fffdabda700 (LWP 16549)]
imp rx null
place part 29043bfd-c944-47cb-ae49-8dc6e1cf7257
tool add part

(horizon-prj-mgr:16493): glibmm-ERROR **: 
unhandled exception (type std::exception) in signal handler:
what: not found

Thread 1 "horizon-prj-mgr" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff3916cd2 in ?? () from /usr/lib/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff3916cd2 in  () at /usr/lib/libglib-2.0.so.0
#1  0x00007ffff3916ecd in g_log_default_handler () at /usr/lib/libglib-2.0.so.0
#2  0x00007ffff39174ef in g_logv () at /usr/lib/libglib-2.0.so.0
#3  0x00007ffff3917680 in g_log () at /usr/lib/libglib-2.0.so.0
#4  0x00007ffff6023852 in Glib::exception_handlers_invoke() () at /usr/lib/libglibmm-2.4.so.1
#5  0x00007ffff6024c44 in  () at /usr/lib/libglibmm-2.4.so.1
#6  0x00007ffff390c0be in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#7  0x00007ffff390df69 in  () at /usr/lib/libglib-2.0.so.0
#8  0x00007ffff390dfae in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#9  0x00007ffff47cd5ae in g_application_run () at /usr/lib/libgio-2.0.so.0
#10 0x000055555559bfc9 in main(int, char**) (argc=1, argv=0x7fffffffdba8) at prj-mgr/prj-mgr-main.cpp:13
```

carrotIndustries, 2017-11-28:
Before starting the application, tell gdb to catch the exception to get a more useful backtrace: 
`catch throw`

Is the pool reasonably up-to-date?

rnestler, 2017-11-28:
> Is the pool reasonably up-to-date?

I did a git pull on the horizon-pool repository just before running the project manager. But maybe i need to regenerate the pool db files?

carrotIndustries, 2017-11-28:
That's very likely to be issue: open the pool in the pool manager and click 'update pool'.

rnestler, 2017-11-28:
Probably more useful backtrace:
```
(gdb) bt
#0  0x00007ffff331bb40 in __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) (obj=obj@entry=0x555555b1e990, tinfo=0x55555583a1e0 <typeinfo for std::runtime_error>, dest=0x7ffff3331680 <std::runtime_error::~runtime_error()>) at /build/gcc-multilib/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:75
#1  0x00005555556c9596 in horizon::Pool::get_filename[abi:cxx11](horizon::ObjectType, horizon::UUID const&) (this=0x555555fbf7e0, type=horizon::ObjectType::PADSTACK, uu=...) at pool/pool.cpp:66
#2  0x00005555556cf47b in horizon::Pool::get_padstack(horizon::UUID const&) (this=this@entry=0x555555fbf7e0, uu=...) at pool/pool.cpp:167
#3  0x00005555556560ef in horizon::Pad::Pad(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Pool&) (this=0x7fffffffbde0, uu=..., j=..., pool=...) at package/pad.cpp:7
#4  0x0000555555652b59 in horizon::Package::Package(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Pool&) (this=0x7fffffffc6b0, uu=..., j=..., pool=...) at pool/package.cpp:154
#5  0x0000555555653ed6 in horizon::Package::new_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, horizon::Pool&) (filename="/home/roughl/proggen/projects/horizon-pool/packages/passive/smd/C0402/package.json", pool=...) at pool/package.cpp:181
#6  0x00005555556cf83a in horizon::Pool::get_package(horizon::UUID const&) (this=this@entry=0x555555fbf7e0, uu=...) at pool/pool.cpp:159
#7  0x00005555555be0d5 in horizon::Part::Part(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Pool&) (this=0x7fffffffd000, uu=..., j=..., pool=...) at pool/part.cpp:29
#8  0x00005555555bfc00 in horizon::Part::new_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, horizon::Pool&) (filename="/home/roughl/proggen/projects/horizon-pool/parts/passive/capacitors/base-0402.json", pool=...) at pool/part.cpp:116
#9  0x00005555556ceebc in horizon::Pool::get_part(horizon::UUID const&) (this=this@entry=0x555555fbf7e0, uu=...) at pool/pool.cpp:177
#10 0x00005555557272cb in horizon::PartBrowserWindow::update_recents() (this=this@entry=0x555555fbf750)
    at prj-mgr/part_browser/part_browser_window.cpp:132
#11 0x000055555572844f in horizon::PartBrowserWindow::placed_part(horizon::UUID const&) (this=0x555555fbf750, uu=...)
    at prj-mgr/part_browser/part_browser_window.cpp:91
#12 0x0000555555715525 in horizon::ProjectManagerAppWindow::handle_req(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&) (this=0x555555c50970, j=...) at prj-mgr/prj-mgr-app_win.cpp:273
#13 0x0000555555716274 in horizon::ProjectManagerAppWindow::<lambda(Glib::IOCondition)>::operator()(Glib::IOCondition) const (__closure=0x555555e7f3e0, cond=<optimized out>) at prj-mgr/prj-mgr-app_win.cpp:254
#14 0x00007ffff6024bef in  () at /usr/lib/libglibmm-2.4.so.1
#15 0x00007ffff390c0be in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#16 0x00007ffff390df69 in  () at /usr/lib/libglib-2.0.so.0
#17 0x00007ffff390dfae in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#18 0x00007ffff47cd5ae in g_application_run () at /usr/lib/libgio-2.0.so.0
#19 0x000055555559bfc9 in main(int, char**) (argc=1, argv=0x7fffffffdba8) at prj-mgr/prj-mgr-main.cpp:13
```

rnestler, 2017-11-28:
> That's very likely to be issue: open the pool in the pool manager and click 'update pool'.

When I try to open the pool in the pool manager now the pool manager seg-faults...

```
Thread 1 "horizon-pool-mg" received signal SIGSEGV, Segmentation fault.
0x00007ffff337b460 in ?? () from /usr/lib/libzmq.so.5
(gdb) bt
#0  0x00007ffff337b460 in  () at /usr/lib/libzmq.so.5
#1  0x00007ffff339a490 in zmq_getsockopt () at /usr/lib/libzmq.so.5
#2  0x00005555557625e7 in zmq::socket_t::getsockopt(int, void*, unsigned long*) const (this=<optimized out>, optvallen_=0x7fffffffbee8, optval_=0x7fffffffbee4, option_=15) at /usr/include/zmq.hpp:564
#3  0x00005555557625e7 in zmq::socket_t::getsockopt<int>(int) const (option_=15, this=<optimized out>) at /usr/include/zmq.hpp:573
#4  0x00005555557625e7 in horizon::PoolNotebook::<lambda(Glib::IOCondition)>::operator() (cond=<optimized out>, __closure=0x555555e6fe40)
    at pool-mgr/pool_notebook.cpp:173
#5  0x00005555557625e7 in sigc::adaptor_functor<horizon::PoolNotebook::PoolNotebook(const string&, horizon::PoolManagerAppWindow*)::<lambda(Glib::IOCondition)> >::operator()<const Glib::IOCondition&> (_A_arg1=<optimized out>, this=0x555555e6fe40)
    at /usr/include/sigc++-2.0/sigc++/adaptors/adaptor_trait.h:89
#6  0x00005555557625e7 in sigc::internal::slot_call1<horizon::PoolNotebook::PoolNotebook(const string&, horizon::PoolManagerAppWindow*)::<lambda(Glib::IOCondition)>, bool, Glib::IOCondition>::call_it(sigc::internal::slot_rep *, sigc::type_trait_take_t) (rep=0x555555e6fe10, a_1=<optimized out>)
    at /usr/include/sigc++-2.0/sigc++/functors/slot.h:148
#7  0x00007ffff6024bef in  () at /usr/lib/libglibmm-2.4.so.1
#8  0x00007ffff390c0be in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#9  0x00007ffff390df69 in  () at /usr/lib/libglib-2.0.so.0
#10 0x00007ffff390ef42 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#11 0x00007ffff5913533 in gtk_native_dialog_run () at /usr/lib/libgtk-3.so.0
#12 0x0000555555760c3f in horizon::PoolManagerAppWindow::handle_open() (this=0x555555a94eb0) at pool-mgr/pool-mgr-app_win.cpp:96
#13 0x00007ffff602a2d9 in Glib::SignalProxyNormal::slot0_void_callback(_GObject*, void*) () at /usr/lib/libglibmm-2.4.so.1
#14 0x00007ffff3bdc6f5 in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#15 0x00007ffff3befeff in  () at /usr/lib/libgobject-2.0.so.0
#16 0x00007ffff3bf4696 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#17 0x00007ffff3bf5920 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#18 0x00007ffff57db2ee in  () at /usr/lib/libgtk-3.so.0
#19 0x00007ffff57db346 in  () at /usr/lib/libgtk-3.so.0
#20 0x00007ffff6fc3b03 in Gtk::Button_Class::released_callback(_GtkButton*) () at /usr/lib/libgtkmm-3.0.so.1
#21 0x00007ffff3bdc6f5 in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#22 0x00007ffff3bf0188 in  () at /usr/lib/libgobject-2.0.so.0
#23 0x00007ffff3bf4696 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#24 0x00007ffff3bf5920 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#25 0x00007ffff57d96a1 in  () at /usr/lib/libgtk-3.so.0
#26 0x00007fffee3981c8 in ffi_call_unix64 () at /usr/lib/libffi.so.6
#27 0x00007fffee397c2a in ffi_call () at /usr/lib/libffi.so.6
#28 0x00007ffff3be627f in g_cclosure_marshal_generic_va () at /usr/lib/libgobject-2.0.so.0
#29 0x00007ffff3bf4c01 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#30 0x00007ffff3bf5920 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#31 0x00007ffff589d024 in  () at /usr/lib/libgtk-3.so.0
#32 0x00007ffff3be5f38 in g_cclosure_marshal_VOID__BOXEDv () at /usr/lib/libgobject-2.0.so.0
#33 0x00007ffff3bf4c01 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#34 0x00007ffff3bf5920 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#35 0x00007ffff589a1cf in  () at /usr/lib/libgtk-3.so.0
#36 0x00007ffff589b84b in  () at /usr/lib/libgtk-3.so.0
#37 0x00007ffff589e5c7 in  () at /usr/lib/libgtk-3.so.0
#38 0x00007ffff5869e34 in gtk_event_controller_handle_event () at /usr/lib/libgtk-3.so.0
#39 0x00007ffff5a39654 in  () at /usr/lib/libgtk-3.so.0
---Type <return> to continue, or q <return> to quit---
#40 0x00007ffff707bee8 in Gtk::Widget::on_button_release_event(_GdkEventButton*) () at /usr/lib/libgtkmm-3.0.so.1
#41 0x00007ffff707e097 in Gtk::Widget_Class::button_release_event_callback(_GtkWidget*, _GdkEventButton*) () at /usr/lib/libgtkmm-3.0.so.1
#42 0x00007ffff58e8488 in  () at /usr/lib/libgtk-3.so.0
#43 0x00007ffff3bf4c01 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#44 0x00007ffff3bf5920 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#45 0x00007ffff5a3b9e5 in  () at /usr/lib/libgtk-3.so.0
#46 0x00007ffff58e51eb in  () at /usr/lib/libgtk-3.so.0
#47 0x00007ffff58e7443 in gtk_main_do_event () at /usr/lib/libgtk-3.so.0
#48 0x00007ffff53ef9f6 in  () at /usr/lib/libgdk-3.so.0
#49 0x00007ffff5422375 in  () at /usr/lib/libgdk-3.so.0
#50 0x00007ffff390c270 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#51 0x00007ffff390df69 in  () at /usr/lib/libglib-2.0.so.0
#52 0x00007ffff390dfae in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#53 0x00007ffff47cd5ae in g_application_run () at /usr/lib/libgio-2.0.so.0
#54 0x00005555555e8549 in main(int, char**) (argc=1, argv=0x7fffffffdba8) at pool-mgr/pool-mgr-main.cpp:12
```

rnestler, 2017-11-28:
After deleting all the `.db` files and running horizon-pool-mgr it seems to work now.

carrotIndustries, 2017-12-26:
closing since we now have logging

closed 2017-12-26 by carrotIndustries