horizon-eda/horizon #23: crash because some error in schematic

steinm, 2017-12-04:
After modifying some symbols in the pool I can't start the schematic editor anymore. I can tell what I precisely did, but maybe this backtrace helps

`Thread 1 "horizon-imp" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  0x00007ffff22cba70 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff22cd19a in __GI_abort () at abort.c:89
#2  0x00007ffff28e3b85 in __gnu_cxx::__verbose_terminate_handler() ()
    at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff28e1956 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff28e19a1 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff28e1be4 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff290a15f in std::__throw_out_of_range(char const*) ()
    at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000000100213aa0 in std::map<horizon::UUID, horizon::SchematicSymbol, std::less<horizon::UUID>, std::allocator<std::pair<horizon::UUID const, horizon::SchematicSymbol> > >::at(horizon::UUID const&) (__k=..., this=<optimized out>) at /usr/include/c++/7/bits/stl_map.h:533
#8  0x0000000100213aa0 in horizon::LineNet::Connection::Connection(nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Sheet*) (this=0x7fffffffcca8, j=..., sheet=<optimized out>) at schematic/line_net.cpp:18
#9  0x000000010021461a in horizon::LineNet::LineNet(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Sheet*) (this=0x7fffffffcc50, uu=..., j=..., sheet=0x7fffffffd180)
    at schematic/line_net.cpp:164
#10 0x000000010020b34d in horizon::Sheet::Sheet(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Block&, horizon::Pool&) (this=0x7fffffffd180, uu=..., j=..., block=..., pool=...)
    at schematic/sheet.cpp:70
#11 0x00000001001fa327 in horizon::Schematic::Schematic(horizon::UUID const&, nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, long, unsigned long, double, std::allocator, nlohmann::adl_serializer> const&, horizon::Block&, horizon::Pool&) (this=0x10085ddb8, uu=..., j=..., iblock=..., pool=...)
    at schematic/schematic.cpp:49
`

carrotIndustries, 2017-12-04:
Looks like this is due to a symbol referenced by a net line missing. Does it crash as well if you checkout a known-good version of the pool?

steinm, 2017-12-04:
It crashes, but with a different error when I use your unmodified pool at github.

`horizon-imp -c /home/steinm/AVR/horizon/attiny-ir-receiver/top_sch.json /home/steinm/AVR/horizon/attiny-ir-receiver/top_block.json
terminate called after throwing an instance of 'std::runtime_error'
  what():  Part 58fd9546-0e88-4fc0-8fac-8a51e3fd4679 not found
Abgebrochen
`
That error msg makes sense, as there are parts in my schematic which are not in the pool at github. I thing it should just handle this somehow without crashing.

steinm, 2017-12-05:
Lukas, can I help somehow with this issue? It currently stops me from using horizon in one of my projects.

carrotIndustries, 2017-12-05:
Mid-term, this will be handled by logging. For a quick-fix, try applying this patch: https://gist.github.com/carrotIndustries/2c2cfae3d066e992433b8e536d97b50e

steinm, 2017-12-06:
That fixes it. It now issues 3 errors when loading the net lines, but at least it starts up again. Thanks.

carrotIndustries, 2017-12-10:
closing since we have logging

closed 2017-12-10 by carrotIndustries