horizon-eda/horizon #25: Compile fails

greenscreenflicker, 2017-12-11:
Compile process fails on Ubuntu.

`g++ -c -I. -Iblock -Iboard -Icommon -Iimp -Ipackage -Ipool -Ischematic -Iutil -Iconstraints -g3 -D_USE_MATH_DEFINES -fdata-sections -ffunction-sections -pthread -std=c++11 -pthread -I/usr/include/uuid -I/usr/include/gtkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib/x86_64-linux-gnu/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib/x86_64-linux-gnu/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/mirclient -I/usr/include/mircore -I/usr/include/mircookie -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib/x86_64-linux-gnu/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 imp/imp_schematic.cpp -o imp/imp_schematic.o
imp/imp_schematic.cpp: In member function ‘void horizon::ImpSchematic::handle_export_pdf()’:
imp/imp_schematic.cpp:330:3: error: ‘GtkFileChooserNative’ was not declared in this scope
   GtkFileChooserNative *native = gtk_file_chooser_native_new ("Save PDF",
   ^
imp/imp_schematic.cpp:330:25: error: ‘native’ was not declared in this scope
   GtkFileChooserNative *native = gtk_file_chooser_native_new ("Save PDF",
                         ^
imp/imp_schematic.cpp:334:13: error: ‘gtk_file_chooser_native_new’ was not declared in this scope
    "_Cancel");
             ^
imp/imp_schematic.cpp:344:54: error: ‘GTK_NATIVE_DIALOG’ was not declared in this scope
   if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                      ^
imp/imp_schematic.cpp:344:55: error: ‘gtk_native_dialog_run’ was not declared in this scope
   if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                       ^
Makefile:454: recipe for target 'imp/imp_schematic.o' failed
make: *** [imp/imp_schematic.o] Error 1
`

greenscreenflicker, 2017-12-11:
Do I have the wrong GTK version installed? GTKFILECHOOSERNATIVE seems not to be supported.
`		GtkFileChooserNative *native = gtk_file_chooser_native_new ("Save PDF",
			GTK_WINDOW(main_window->gobj()),
			GTK_FILE_CHOOSER_ACTION_SAVE,
			"_Open",
			"_Cancel");
		auto chooser = Glib::wrap(GTK_FILE_CHOOSER(native));
		chooser->set_do_overwrite_confirmation(true);
		if(last_pdf_filename.size()) {
			chooser->set_filename(last_pdf_filename);
		}
		else {
			chooser->set_current_name("schematic.pdf");
		}

		if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
			std::string fn = chooser->get_filename();
			last_pdf_filename = fn;
			export_pdf(fn, *core.c->get_schematic(), core.r);
		}`

greenscreenflicker, 2017-12-11:
Next thing:
dialogs/dialogs.cpp:364:47: error: converting to ‘std::tuple<bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double>’ from initializer list would use explicit constructor ‘constexpr std::tuple< <template-parameter-1-1> >::tuple(_UElements&& ...) [with _UElements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, long int&, double&}; <template-parameter-2-2> = void; _Elements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double}]’
    return {true, filename, layer, width, scale};
                                               ^
dialogs/dialogs.cpp:367:48: error: converting to ‘std::tuple<bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double>’ from initializer list would use explicit constructor ‘constexpr std::tuple< <template-parameter-1-1> >::tuple(_UElements&& ...) [with _UElements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, long int&, double&}; <template-parameter-2-2> = void; _Elements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double}]’
    return {false, filename, layer, width, scale};

greenscreenflicker, 2017-12-11:
next:
property_panels/property_panel.cpp: In static member function ‘static horizon::PropertyPanel* horizon::PropertyPanel::create(horizon::ObjectType, horizon::Core*, horizon::PropertyPanels*)’:
property_panels/property_panel.cpp:121:49: error: no matching function for call to ‘Gtk::Builder::get_widget_derived(const char [14], horizon::PropertyPanel*&, horizon::ObjectType&, horizon::Core*&)’
   x->get_widget_derived("PropertyPanel", w, t, c);

greenscreenflicker, 2017-12-11:
imp/fab_output_window.cpp:75:4: error: ‘GtkFileChooserNative’ was not declared in this scope
    GtkFileChooserNative *native = gtk_file_chooser_native_new ("Select output directory",
    ^
imp/fab_output_window.cpp:75:26: error: ‘native’ was not declared in this scope
    GtkFileChooserNative *native = gtk_file_chooser_native_new ("Select output directory",
                          ^
imp/fab_output_window.cpp:79:14: error: ‘gtk_file_chooser_native_new’ was not declared in this scope
     "_Cancel");
              ^
imp/fab_output_window.cpp:84:55: error: ‘GTK_NATIVE_DIALOG’ was not declared in this scope
    if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                       ^
imp/fab_output_window.cpp:84:56: error: ‘gtk_native_dialog_run’ was not declared in this scope
    if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                        ^
imp/fab_output_window.cpp: In static member function ‘static horizon::FabOutputWindow* horizon::FabOutputWindow::create(Gtk::Window*, horizon::Board*, horizon::FabOutputSettings*)’:
imp/fab_output_window.cpp:156:42: error: no matching function for call to ‘Gtk::Builder::get_widget_derived(const char [7], horizon::FabOutputWindow*&, horizon::Board*&, horizon::FabOutputSettings*&)’
   x->get_widget_derived("window", w, b, s);

carrotIndustries, 2017-12-11:
Which ubuntu version are you using? GtkFileChooserNative requires at least Gtk 3.20, i.e. at least ubuntu 17.04. The other errors are probably due too an older version of gcc not being as smart as I'm hoping for. Since the next LTS is only a few months away, fixing these issues isn't worthwhile from my point. I'm happy to accept patches though.

rnestler, 2017-12-12:
@carrotIndustries Maybe we should link in the README to the [build instructions in the wiki](https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Linux). Also maybe state in the wiki on which distro you develop (i.e. the one that will always work) and which minimal Ubuntu version is required.

greenscreenflicker, 2017-12-12:
@carrotIndustries @rnestler I'm using Ubuntu 16.04 LTS.
GTK version is 3.18.9-1ubuntu3.3 - so thats why its not supported
GCC version is gcc version 5.4.0 20160609 

From my personal perspective the systems setup isn't too old. However, when the new LTS is there i will setup a new PC, so then I can try your software. Would have loved to try it right now, but anway...

From my personal point you can close this issue, as the build instructions have been updated.

closed 2017-12-12 by carrotIndustries