horizon-eda/horizon #27: Building under Windus using msys

greenscreenflicker, 2017-12-13:
I followed the wiki to try to compile, however I do not know msys.
Packages are updated, however, there seem to miss a few. 
`
LTIPC2082+Michael@LTIPC2082 MSYS ~/horizon
$ make
Package gio-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gio-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gio-2.0' found
/bin/sh: --: Ungültige Option
Aufruf: /bin/sh [Lange GNU Option] [Option] ...
                /bin/sh [Lange GNU Option] [Option] Script-Datei ...
Lange GNU Optionen:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --protected
        --rcfile
        --restricted
        --verbose
        --version
        --wordexp
Shell-Optionen:
        -ilrsD oder -c Kommando oder -O shopt_option            (Nur Aufruf)
        -abefhkmnptuvxBCHP oder Option -o
Package sqlite3 was not found in the pkg-config search path.
Perhaps you should add the directory containing `sqlite3.pc'
to the PKG_CONFIG_PATH environment variable
No package 'sqlite3' found
Package yaml-cpp was not found in the pkg-config search path.
Perhaps you should add the directory containing `yaml-cpp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'yaml-cpp' found
Package gtkmm-3.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gtkmm-3.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gtkmm-3.0' found
Package epoxy was not found in the pkg-config search path.
Perhaps you should add the directory containing `epoxy.pc'
to the PKG_CONFIG_PATH environment variable
No package 'epoxy' found
Package cairomm-pdf-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `cairomm-pdf-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'cairomm-pdf-1.0' found
Package librsvg-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `librsvg-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'librsvg-2.0' found
Package libzmq was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzmq.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzmq' found
g++ -c -I. -Iblock -Iboard -Icommon -Iimp -Ipackage -Ipool -Ischematic -Iutil -Iconstraints -g3 -D_USE_MATH_DEFINES -DWIN32_UUID -fdata-sections -ffunction-sections  -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 util/uuid.cpp -o util/uuid.o
make: g++: Kommando nicht gefunden
make: *** [Makefile:454: util/uuid.o] Fehler 127
`

Please support what to do. I seem to be quite unlucky compiling.

carrotIndustries, 2017-12-13:
Make sure you're using the mingw environment, not the msys one: https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Windows#start-msys-console

greenscreenflicker, 2017-12-15:
[image]
File size is huge, and still files are missing. What am I doing wrong?

carrotIndustries, 2017-12-15:
The huge file size is due to debug symbols. You'll have to run the executables directly from the mingw shell or use the make_bindist.sh script to create a folder with the DLLs in it so you have something to double-click on.

greenscreenflicker, 2017-12-18:
Before running the script, it did not work from the console. So i ran the script, now it runs from the console, however not from windows. With the same lib warnings as before.
The binaries you provide run without any issue. Compiling now kinda works.

Thank you again for developping, keep up the good work!

I will try to make a simple PCB to you pcb program out and document my work on mikrocontroller.net then.

closed 2017-12-18 by greenscreenflicker