horizon-eda/horizon #28: New part - Bring window to front

greenscreenflicker, 2017-12-18:
It would be convinient when clicking on "place part" would bring the part window to front.

carrotIndustries, 2017-12-18:
Based on your previous issue, I'm guessing you're using windows.
Actually I'm trying to do show the window. Does the window light up in the taskbar?

greenscreenflicker, 2017-12-19:
@carrotIndustries : Yes, windows.
The window blinks in the task bar, but does not come to front.
Blinking in taskbar, see img below.
[image]

carrotIndustries, 2017-12-19:
Unfortunately, that's a limitation arising from the fact that the schematic editor and the project manager that the part browser window belongs to are separate processes. To protect the user from nasty applications, Windows doesn't allow applications other than the one with input focus to bring windows to the front. (At least, that's how I understood it)

closed 2017-12-19 by carrotIndustries

greenscreenflicker, 2017-12-20:
Just found out this: Pressing pp (Place part) puts the window to the front and works. However clicking on the button, does not put it to front.

greenscreenflicker, 2017-12-28:
@carrotIndustries : New discovery: The window is already there and is therefore only highlighted. 
If a part is placed, this window should close automatically. Then, when opening a new window, the window comes to front.