horizon-eda/horizon #33: Package editor: Line properties

greenscreenflicker, 2017-12-26:
It would be handy to have a start and a end position directly modifyable using a text entry.
[image]
Second, it would be nice to have a default line width.

See the pic for a rough scetch.

atoav, 2018-02-01:
I agree to the ability to set default line widths (or maybe just remembering the last selected would work).

For the Positions I think `Enter Datum` could do what you want? 
[image]
Having it in the sidebar makes probably also sense (there is enough space, you could display the values of all the junctions at once, etc)

carrotIndustries, 2018-02-01:
> I agree to the ability to set default line widths (or maybe just remembering the last selected would work).

Currently Tools can't store any state between invocations, but being able to do so is on my todo list as well. Being able to set a default line width would be a byproduct of saving state.