horizon-eda/horizon #35: Pool manager: Does not rember last opened pool

greenscreenflicker, 2017-12-26:
Restarting the pool manager does not rember the old used pools. As you like to have only one pool it would be best if you would start directly into the pool.

Also it would be nice to have a button "get pool" and "update pool" to get it from git.

[image]

carrotIndustries, 2017-12-26:
The download button is there in the latest windows release, an improved recent chooser is WIP right now.

closed 2017-12-26 by carrotIndustries

greenscreenflicker, 2017-12-28:
@carrotIndustries : Yes, i now got the download option and the update option after updating the software. However, the pool manager does not remember the last used pools.
[image]

carrotIndustries, 2017-12-29:
Fixed in https://github.com/carrotIndustries/horizon/commit/c53bae290fc869f0eef677f1e5fb352cf071daec