horizon-eda/horizon #36: Pool manager: Deleting non-existing file does not remove it from pool.

greenscreenflicker, 2017-12-28:
[image]
If a file does not exist, however, its in the padstack listed, and you want to delete the non-existing file, it stays in the pool.

greenscreenflicker, 2017-12-28:
Current workaround is to copy a file with the name, and then the delete works.

carrotIndustries, 2017-12-28:
I don't see your issue: After deleting an item using the pool manager, the pool gets updated so, the deleted item won't be included in the pool anymore. When manually deleting an item, you'll need to run the pool update manually for the pool manager to pick it up.

greenscreenflicker, 2018-01-01:
@carrotIndustries: I was editing in the file system, not in the pool manager. What i propose, is to remove the file from the database when it's not existing on the file system.  
When deleteting (when non-existent on the fs), to remove it from the database in the pool manager. So basically if the delete on the file system fails, it should remove it from the database.

carrotIndustries, 2018-01-01:
In fact there's no such thing as "removing from the database". Every time you click "Update pool" the entire database is cleared and rebuilt from the files found on disk. So just delete the file however you like, update the pool and you're good to go.

greenscreenflicker, 2018-01-02:
Okay, got it.

closed 2018-01-02 by greenscreenflicker