horizon-eda/horizon #44: Pool manager: Downloading pools fails

greenscreenflicker, 2018-01-01:
[image]
See image...

carrotIndustries, 2018-01-01:
Did you use the zip or did you build it yourself? In the latter case, you'll have to copy the cert bundle to the application directory. I updated the documentation to mention this: https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Windows#running

When reporting issues, please specify which build you're using. It's shown in the pool/project manger start page.

closed 2018-01-01 by carrotIndustries

greenscreenflicker, 2018-01-02:
@carrotIndustries : I propose to add the copy instructions to ./make_bindings.sh for an improved user experince. (And yes, building myself.)

carrotIndustries, 2018-01-02:
The `make_bindist.sh` script already copies the certificates to the output directory.

greenscreenflicker, 2018-01-04:
@carrotIndustries : I see it in the script too, i agree. However, it does not work for me in the script. Maybe you can test it again, plz?
If i copy the file manually, it works. So the cp instruction seems to fail. Maybe you can debug that...