horizon-eda/horizon #46: Compile fails in current version

greenscreenflicker, 2018-01-02:
Compile fails in 9f8fa5b
[image]

**All libs are up to date**

carrotIndustries, 2018-01-02:
Haven't seen this before. You may try doing a full rebuild by issuing `make clean` before building.

greenscreenflicker, 2018-01-02:
@carrotIndustries : I did a clean before. Doesn't solve it.

carrotIndustries, 2018-01-02:
This is due to the mingw people forgetting to recompile yaml-cpp after a boost update. When rebuilding yaml-cpp, the issue goes away. Unfortunately there's an issue with the latest yaml-cpp that needs to be sorted out before rebuilding the package: https://github.com/jbeder/yaml-cpp/issues/551

greenscreenflicker, 2018-01-02:
Thanks for sorting out.

closed 2018-01-02 by greenscreenflicker