horizon-eda/horizon #50: Pool manager: Custom padstack does not work in footprint generator

greenscreenflicker, 2018-01-04:
I created a custom padstack with round edges for smd components. Now using the dual footprintgenerator, the old obround (on which my design is based on) is used. 

Where is my problem? What am I doing wrong? Plz support.
[smd-half-round.zip](https://github.com/carrotIndustries/horizon/files/1603586/smd-half-round.zip)

carrotIndustries, 2018-01-04:
This is due to the parameter program rearranging the shapes. You'll have to adjust the parameter program as well to suit your new padstack. The padstack you want to create can be created much easier by using the obround shape.

greenscreenflicker, 2018-01-04:
@carrotIndustries : It's a bit difficult to me to follow:  You mean one would have to change the source code that the new padstack works? You mean that? Please explain slowly for me.

carrotIndustries, 2018-01-04:
No, each padstack comes with a small program that takes care of modifying the padstack according to the parameters, i.e. moving shapes around and adjusting their size: https://github.com/carrotIndustries/horizon/wiki/Parameter-programs To edit the program, click the "Parameters..." button in the padstack editor. Then click "Apply" for running the program.

greenscreenflicker, 2018-01-04:
Puh, to be honest, it looks very difficult to me, to create rounded SMD edges. 
Could it be possible that you add a rounded SMD pad-option to the pool?
Rounded SMD pads have better reflow paste characteristics and are therefore prefered by me.

carrotIndustries, 2018-01-04:
https://github.com/carrotIndustries/horizon-pool/commit/176544b299b56e05b3dd5dea1f858e14a10c041d adds this padstack you'll need at leas https://github.com/carrotIndustries/horizon/commit/0a2210f28736d49ecb1dc60a55271c24c3174e3d for this padstack to work

closed 2018-01-04 by carrotIndustries

greenscreenflicker, 2018-01-05:
@carrotIndustries  Works so great! Thanks for the implementation.