horizon-eda/horizon #55: Package editor: Rule checker

greenscreenflicker, 2018-01-05:
I'd propose to add a rule checker when creating packages. As many people should later create packages, "wildwuchs" (deutsch: rank-grove) would result, as everybody would do things slightly diffrently.

The package checker could do e.g. those things:
1. Check if a name for the part is given
2. Check if a $RD tag is on the footprint
3. Check if $RD tag has the same size arrording to the standard (0.15mm line width - 1mm height)
4. Check if Countyard is present
5. Check if silkscreen is present
... This could be extended, when more using it, more ideas will come...

Only if this DRC checks are passed, a merge to the repository should be allowed.

carrotIndustries, 2018-01-05:
https://github.com/carrotIndustries/horizon/commit/6e49cb9e8d11541f534cf21c770ab515ec5e051a adds some of these. New checks should be fairly trivial to add: https://github.com/carrotIndustries/horizon/blob/master/package/package_rules_check.cpp

greenscreenflicker, 2018-01-06:
@carrotIndustries 
This error message is not quite clear to me. Maybe you add a proposed "fixed" to the error msg.
[image]

carrotIndustries, 2018-01-06:
Oh, i got that text wrong (fixed in https://github.com/carrotIndustries/horizon/commit/3b35e7521456b659c27a06222aa99385298940c8), it should read assembly reference designator. That's the one you're missing: https://github.com/carrotIndustries/horizon/wiki/Creating-a-Package#assembly-outline

greenscreenflicker, 2018-01-12:
@carrotIndustries Ok, clear now. 

Can you add the following more test:
1. Check if package outline is present
2. Check if Assembly outline is present
3. (Maybe this exists) Check if Countyard is present
4. Check that package outline fits in countyard
5. Check that assembly outline fits in countyard

greenscreenflicker, 2018-01-17:
@carrotIndustries : One more rule to check:
Only one countyard polygon inside the footprint.

greenscreenflicker, 2018-02-23:
@carrotIndustries As you are so crictical about submitting to the horizon pool, i think strict rule checking allows the reduction of errors.

pwnorbitals, 2021-02-09:
Can be closed ?