horizon-eda/horizon #57: New logo?

greenscreenflicker, 2018-01-05:
I'd propose the idea of a new logo for the program, as it advanches so fast, a new logo would look more professional. I was playing with logo maker for a minute, and they seem to have good logos to offer (as far as i see for free...)
[image]
https://www.logomaker.com/

Of cause, it will not be as good as a designed one, however, i think for the moment it would look quite nice...

carrotIndustries, 2018-01-06:
Thanks for the effort, but no. In case any designer reads along, the current 80s-themed logo is approximately what I'd like to have. Some more inspiration: https://www.google.de/search?tbm=isch&q=80s+math+book+cover&chips=q:80s+math+book+cover,online_chips:vaporwave

atoav, 2018-01-29:
I happen to be a graphic designer and want to offer myself to create a nice logo for horizon.
Logos typically are a very subjective topic, but I think one can boil down a few objective aspects ( considering a icon for a open source project):

1. a logo should fit the project (or its name)
2. a logo should offer a distinctive and harmonic color to make it recognizeable
3. a icon should work especially well at both big and small sizes
4. a icon should work on bright and black backgrounds alike

From point 3 rationally follows, that the icons complexity has to be low enough to yield a working icon at a small size. This is especially true for the current icon. 

I made a single draft – for paying customers I'd typically do more than one but I don't know wether this is even wanted here. I kinda liked to contribute, because of all the open source EDA projects horizon looks most promising to me.  Typically I'd start out with a black and white shape version and start tuning in colors later. The small icons will be drawn pixel by pixel to maximize small scale readability.  Obviously I'd need to be sure that a certain draft is to be taken, to avoid loops and unecessary work.  

In this draft I tried to catch the horizion name, like you did in your original icon, while simplyfing it and still keeping some 80s spirit alive (I hope it survived : ). Note that the different backgrounds are just for display purposes. 

[image]
[image]
[image]
[image]
[image]

Feel free to critique etc.

carrotIndustries, 2018-01-30:
Awesome! Nice to have something different than the usual variation of the "Board/Schematic/Gate/IC with or without pen" theme.

atoav, 2018-01-30:
I'll play around with it for a bit and report back if anything better comes up. Any preferences on the colours?

atoav, 2018-01-30:
I made two new monochrome versions, that sort of build up on the old draft. Both try to be a bit more distinctive, than the more generic looking drafts from yesterday. I tried to keep the horizon metaphor, but extend it with a "planetary" aspect. After all, electronics are the most powerful thing we as humanity ever came up with.  

At the same time this draft is abstract enough to be literally everything else. It could be a single pad on a PCB looked at from above or a swimming pool seen from the top of a tall building.  

| Variant | Image |
| -------- |:-------:|
| Black Sun (1080px) | [image] |
| Black Sun (64px) | [image] |
| White Moon (1080px) | [image] |
| White Moon (64px) | [image] |

carrotIndustries, 2018-01-30:
Mind that the logo will end up as 16x16px icon in the the top left of the application window as in https://raw.githubusercontent.com/wiki/carrotIndustries/horizon/images/pool-mgr.png and probably people's task bars/panels. To my judgment, the 'white moon' one stays the most recognizable.

In terms of colors, i like your first idea, but feel free to experiment.

greenscreenflicker, 2018-02-07:
@atoav @carrotIndustries For me the moon logo looks just amazing. I love it. I think it's special and makes it recognizable.

I played a little bit with some colors, I would like. I must denote, I'm everything else then a designer, so it's just a rough idea.
[image]

atoav, 2018-02-10:
I'd avoid gradients for well defined multiple colors. I changed the spacing between the lines on the white_moon draft a little bit and added colors. While playing around with some sunset variation I thought: "Hey – who says that it is a horizon seen on earth?" So I imagined the circle as a pale blue dot. Well.
| Variant | Image |
|:-------:|:--------:|
| Pale Blue Dot Mars (1080px) | [image] |
| Pale Blue Dot Mars (16px) | [image] |  

I kinda like how the small Icon still stays recognizable, the colors represent Mars but could give it a little to much 70s feel...  

So I tried another variant for the moon:  
| Variant | Image |
|:-------:|:--------:|
| Pale Blue Dot Moon (1080px) | [image] |
| Pale Blue Dot Moon (16px) | [image] |  
Despite the lack of recognizable hue in the "landscape" part of things, I think this draft still stays very recognizable at small sizes, color-wise. For the sacrifice in color we get a little more serious professionalism.  

What do you think?

carrotIndustries, 2018-02-11:
Great work! 

> What do you think?

The not-so-much colored one indeed looks more professional, let's stick with that one.

atoav, 2018-02-11:
I created a repo called [horizon-logo](https://github.com/atoav/horizon-logo) with all possible source formats and exports. The PNGs with the resolutions 16px, 32px and 64px have been made pixel perfect and should look decent at small sizes now. Feel free to use it within horizon.

carrotIndustries, 2018-02-11:
b805a35ad5b55b384da64fccf31c9f24d9c32016 adds the new icons

closed 2018-02-11 by carrotIndustries

greenscreenflicker, 2018-02-12:
Looks amazing! @atoav good work. However, i see there some artefacts, the circle is not round.
[image]

atoav, 2018-02-12:
Interesting over here it looks like this with a small taskbar:  
[image]  

and with a big Taskbar:  
[image]  

As you can see I don't have such scaling artifacts and they are certainly not in the images themselves. I don't know how the Windows Taskbar handles Icon scaling and if there are any ways to optimize it (from the position of a graphic designer)