horizon-eda/horizon #58: Part-specific 3D models

fruchti, 2018-01-10:
I think 3D models should be assigned to parts rather than packages. There are many cases of parts sharing the same footprint which should have different 3D models, for example:
- Parts with different possible heights, like electrolytic capacitors or pin headers
- Parts in different colours (like LEDs)

Creating the same package twice for these cases seems rather redundant to me.

Instead of setting the 3D model directly for a part, the package could also have a list of alternative 3D models, from which one can be chosen for each part.

carrotIndustries, 2018-01-10:
You're the second one to comment on this, so seems like that's a thing.

> Instead of setting the 3D model directly for a part, the package could also have a list of alternative 3D models, from which one can be chosen for each part.

That's probably what I'll implement, since that way on can see the 3D models in conjunction with the footprint.

greenscreenflicker, 2018-01-13:
@carrotIndustries I agree with @fruchti viewpoint.

commit b8e6bd3f (2018-01-14): packages now have multiple 3d models

parts can select from these models

fixes #58
closed 2018-01-14 by carrotIndustries