horizon-eda/horizon #61: Package editor: Add rotating option to step files

greenscreenflicker, 2018-01-12:
[image]
Option like, x,y,z offset and rotation required for using step models.

By the way: Where should the step models be saved?

carrotIndustries, 2018-01-12:
I didn't add such an option since specifying offset/rotation by typing in a number isn't exactly the way it should be done. To me, there are primarily two sources of 3D Models:
 - The Kicad 3D models library at https://github.com/KiCad/kicad-packages3D/
 - 3D models provided by vendors

The kicad 3d models already have the canonical zero orientation, so there's no need to rotate these. Models provided by vendors are usually all over the place requiring reorientation with a proper MCAD tool such as FreeCAD.

See the existing 3d models for guidance on where to put the files.

greenscreenflicker, 2018-01-13:
@carrotIndustries : It seems, like the 3d models are not copied by cloning the pool.  Maybe you can fix this?

@carrotIndustries: I will try to use kicad packages as first resource. The database seems quite good to me.

carrotIndustries, 2018-01-13:
Yeah, integration with the download/upgrade things in the pool manager is still WIP.

carrotIndustries, 2018-01-15:
since 2953d2f8669ff40b25e23bc0326fe11341a6be43 the pool manager handle 3d models in download and upgrade

closed 2018-01-15 by carrotIndustries

greenscreenflicker, 2018-01-17:
@carrotIndustries There is some commenting on this issue on Mikrocontroller.net
https://www.mikrocontroller.net/topic/417908#5282789