horizon-eda/horizon #62: Package editor: Infinite loop when deleting a package. (Kind of Crash)

greenscreenflicker, 2018-01-12:
When deleting a package in package editor the current version 72afa3e there seems to be an infinite loop; the program does not terminate.

greenscreenflicker, 2018-01-12:
There was an alternative package for it defined (the one i was deleting), maybe this is a hint.

greenscreenflicker, 2018-01-12:
Ok, found a manual solution: Setting "alternate_for": "[uid]", setting it to nothing solves it.
So maybe the delete function should check if the package deleted is defined as an alternative package, and if yes, remove that tag. 
The update function should not while(1) if a tag cannot be found.

carrotIndustries, 2018-01-12:
The issue is related to the way the pool manager handles updating the packages: It repeatedly crawls the packages directory until all references are resolved. When a package can't be found, this leads to the infinite loop you're describing.

To fix this properly, the pool manger needs to figure out all dependencies beforehand, build a dependency graph and error out  if there are unresolved dependencies or cycles.

commit c30a2440 (2018-01-21): make pool update more robust

fixes #62
closed 2018-01-21 by carrotIndustries