horizon-eda/horizon #64: No courtyard found

greenscreenflicker, 2018-01-13:
[image]
I get the following error message. Maybe you can explain the msg to me.

I attach you the footprint.
[sot23-5-rect.zip](https://github.com/carrotIndustries/horizon/files/1628266/sot23-5-rect.zip)

carrotIndustries, 2018-01-13:
The file you attached doesn't contain a courtyard polygon. Add one that matches the specifications given in the error message.

greenscreenflicker, 2018-01-13:
@carrotIndustries  It's weired, look at the image above, there's a courtyard polygon.  Sorry, I'm afarid i just do not understand.

carrotIndustries, 2018-01-15:
The file you sent doesn't contain one. Maybe you attached the wrong one?

greenscreenflicker, 2018-01-17:
Okay, problem solved: I had serveral countyards in the design, but one rectange was only one line. So deleting this, solved everything. 
For the rulechecker there should be an option that only one countyard layer is inside.

closed 2018-01-17 by greenscreenflicker