horizon-eda/horizon #66: DXF import in the board editor

fruchti, 2018-01-16:
It would be handy to be able to directly import a DXF file for the outline layer.

carrotIndustries, 2018-01-16:
The tool is already there, just hit space and search for dxf. b319bdfe210cfaf0e6edc06845db626d58290603 adds it to the hamburger menu for improved discoverability.

Anyhow, this tool imports the dxf file as lines, whereas the outline has to be defined as a polygon. To make this tool useful for your purpose, we'd need another tool for converting line loops into polygons. Pull requests are welcome :)

fruchti, 2018-07-22:
It would be great if the line loop to polygon tool would work for arcs, too. Then, this issue would be resolved.

carrotIndustries, 2018-07-23:
e6d524de3d362bf7d82861d209084ab7943d36f4 enables this, and 78de1aae7dca90fc2528fd2146ba8fd071c69cab makes polygons always render arcs, even if the two radii aren't the same

closed 2018-07-23 by carrotIndustries

fruchti, 2018-07-23:
Thanks a lot! It's working perfectly now.