horizon-eda/horizon #69: Interested in macosx support?

jacobrosenthal, 2018-01-20:
Seem to have found most deps in homebrew with 
`brew install yaml-cpp sqlite3 librsvg cairomm libepoxy gtkmm3 ossp-uuid boost zeromq glm libgit2 opencascade`

and seem to need 	`#include <uuid/uuid.h>` instead of `<uuid.h>` in uuid.hpp

but Im stuck here:
```
$ make
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -fdata-sections -ffunction-sections -D_REENTRANT -I/usr/local/Cellar/yaml-cpp/0.5.3/include -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/usr/local/Cellar/gtkmm3/3.22.2/include/gtkmm-3.0 -I/usr/local/Cellar/gtkmm3/3.22.2/lib/gtkmm-3.0/include -I/usr/local/Cellar/atkmm/2.24.2/include/atkmm-1.6 -I/usr/local/Cellar/gtk+3/3.22.26/include/gtk-3.0/unix-print -I/usr/local/Cellar/gtkmm3/3.22.2/include/gdkmm-3.0 -I/usr/local/Cellar/gtkmm3/3.22.2/lib/gdkmm-3.0/include -I/usr/local/Cellar/glibmm/2.54.1/include/giomm-2.4 -I/usr/local/Cellar/glibmm/2.54.1/lib/giomm-2.4/include -I/usr/local/Cellar/pangomm/2.40.1/include/pangomm-1.4 -I/usr/local/Cellar/pangomm/2.40.1/lib/pangomm-1.4/include -I/usr/local/Cellar/glibmm/2.54.1/include/glibmm-2.4 -I/usr/local/Cellar/glibmm/2.54.1/lib/glibmm-2.4/include -I/usr/local/Cellar/gtk+3/3.22.26/include/gtk-3.0 -I/usr/local/Cellar/glib/2.54.3/include/gio-unix-2.0/ -I/usr/local/Cellar/cairo/1.14.12/include/cairo -I/usr/local/Cellar/pango/1.40.14/include/pango-1.0 -I/usr/local/Cellar/harfbuzz/1.7.4/include/harfbuzz -I/usr/local/Cellar/graphite2/1.3.10/include -I/usr/local/Cellar/pango/1.40.14/include/pango-1.0 -I/usr/local/Cellar/atk/2.26.1/include/atk-1.0 -I/usr/local/Cellar/cairo/1.14.12/include/cairo -I/usr/local/Cellar/libepoxy/1.4.3/include -I/usr/local/Cellar/cairomm/1.12.2/include/cairomm-1.0 -I/usr/local/Cellar/cairomm/1.12.2/lib/cairomm-1.0/include -I/usr/local/Cellar/libsigc++/2.10.0/include/sigc++-2.0 -I/usr/local/Cellar/libsigc++/2.10.0/lib/sigc++-2.0/include -I/usr/local/Cellar/cairo/1.14.12/include/cairo -I/usr/local/Cellar/librsvg/2.42.0/include/librsvg-2.0 -I/usr/local/Cellar/gdk-pixbuf/2.36.11/include/gdk-pixbuf-2.0 -I/usr/local/Cellar/libpng/1.6.34/include/libpng16 -I/usr/local/Cellar/cairo/1.14.12/include/cairo -I/usr/local/Cellar/glib/2.54.3/include/glib-2.0 -I/usr/local/Cellar/glib/2.54.3/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.41/include -I/usr/local/Cellar/pixman/0.34.0_1/include/pixman-1 -I/usr/local/Cellar/fontconfig/2.12.6/include -I/usr/local/opt/freetype/include/freetype2 -I/usr/local/Cellar/libpng/1.6.34/include/libpng16 -I/usr/local/Cellar/zeromq/4.2.3/include -I/usr/local/Cellar/libgit2/0.26.0/include -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/pool/symbol.cpp -o src/pool/symbol.o
In file included from src/pool/symbol.cpp:1:
In file included from src/pool/symbol.hpp:4:
src/common/common.hpp:105:48: warning: suggest braces around initialization of subobject [-Wmissing-braces]
                        std::array<T, 2> as_array () const {return {x, y};}
                                                                    ^~~~
                                                                    {   }
src/pool/symbol.cpp:99:28: note: in instantiation of member function 'horizon::Coord<long long>::as_array' requested here
                j["position"] = position.as_array();
                                         ^
In file included from src/pool/symbol.cpp:1:
In file included from src/pool/symbol.hpp:3:
In file included from 3rd_party/json.hpp:50:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/map:631:15: error: object of type
      'std::__1::pair<horizon::UUID, horizon::SymbolPin>' cannot be assigned because its copy assignment operator is implicitly deleted
        {__nc = __v.__cc; return *this;}
              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tree:1620:35: note: in instantiation of member function
      'std::__1::__value_type<horizon::UUID, horizon::SymbolPin>::operator=' requested here
                __cache->__value_ = *__first;
                                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tree:1550:9: note: in instantiation of function template specialization
      'std::__1::__tree<std::__1::__value_type<horizon::UUID, horizon::SymbolPin>, std::__1::__map_value_compare<horizon::UUID, std::__1::__value_type<horizon::UUID,
      horizon::SymbolPin>, std::__1::less<horizon::UUID>, true>, std::__1::allocator<std::__1::__value_type<horizon::UUID, horizon::SymbolPin> >
      >::__assign_multi<std::__1::__tree_const_iterator<std::__1::__value_type<horizon::UUID, horizon::SymbolPin>,
      std::__1::__tree_node<std::__1::__value_type<horizon::UUID, horizon::SymbolPin>, void *> *, long> >' requested here
        __assign_multi(__t.begin(), __t.end());
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/map:914:21: note: in instantiation of member function
      'std::__1::__tree<std::__1::__value_type<horizon::UUID, horizon::SymbolPin>, std::__1::__map_value_compare<horizon::UUID, std::__1::__value_type<horizon::UUID,
      horizon::SymbolPin>, std::__1::less<horizon::UUID>, true>, std::__1::allocator<std::__1::__value_type<horizon::UUID, horizon::SymbolPin> > >::operator=' requested
      here
            __tree_ = __m.__tree_;
                    ^
src/pool/symbol.cpp:233:8: note: in instantiation of member function 'std::__1::map<horizon::UUID, horizon::SymbolPin, std::__1::less<horizon::UUID>,
      std::__1::allocator<std::__1::pair<const horizon::UUID, horizon::SymbolPin> > >::operator=' requested here
                pins = sym.pins;
                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/utility:322:5: note: copy assignment operator is implicitly deleted
      because 'pair<horizon::UUID, horizon::SymbolPin>' has a user-declared move constructor
    pair(pair&&) = default;
    ^
1 warning and 1 error generated.
make: *** [src/pool/symbol.o] Error 1
$ 
```

carrotIndustries, 2018-01-20:
I'm afraid, but porting horizon to OS X is futile until Gtk's Quartz backend supports OpenGL since OpenGL is required for rendering everything from symbol to board.

endofexclusive, 2018-01-20:
For reference, the same compilation error appears on FreeBSD 11.
```
$ uname -s -r -m
FreeBSD 11.1-RELEASE amd64
$ c++ -v
FreeBSD clang version 4.0.0 (tags/RELEASE_400/final 297347) (based on LLVM 4.0.0)
Target: x86_64-unknown-freebsd11.1
Thread model: posix
InstalledDir: /usr/bin
$ gmake clean
$ gmake -k > log 2>&1
$ grep "cannot be assigned because its copy assignment operator is implicitly deleted" log
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::SymbolPin>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::LineNet>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::NetLabel>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::PowerSymbol>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::BusLabel>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/map:631:15: error: object of type 'std::__1::pair<horizon::UUID, horizon::BusRipper>' cannot be assigned because its copy assignment operator is implicitly deleted
/usr/include/c++/v1/__tree:1620:35: error: object of type 'horizon::SelectableRef' cannot be assigned because its copy assignment operator is implicitly deleted
```

carrotIndustries, 2018-01-20:
TBH, I don't see what's wrong with that code since it compiles just fine with clang 5.0.1. Anyone got a clue?

korken89, 2018-01-20:
Seem like the `horizon::UUID` and `horizon::SelectableRef` has a deleted copy assignment constructor. Check the source for them. And if they are external, it's probably a version difference of external packages.

carrotIndustries, 2018-01-22:
As far as I can tell, neither of them have a in any way special constructor/overloaded assignment operator. Looks like the compiler/libc got smarter...

Instead of assigning to the map, try this:

    pins.clear();
    pins.insert(sym.pins.begin(), sym.pins.end());

ozel, 2018-02-08:
re Quartz/Gtk: can you comment on where exactly code is missing or if in some quartz related repo opengl support is currently beeing worked on?

found this by watching the FOSDEM talks, BTW. Great work!

carrotIndustries, 2018-02-08:
As soon as the gnome folks sort out their infrastructure hiccups, see https://bugzilla.gnome.org/show_bug.cgi?id=740199

carrotIndustries, 2018-02-23:
closing since mac os x support will only arrive once GtkGLArea supports quartz

closed 2018-02-23 by carrotIndustries