horizon-eda/horizon #70: Build system + Warnings are not errors + General recommendations

korken89, 2018-01-22:
Hi, cool that you are starting a new EDA tool without legacy!

I had a look at the build system and compile outputs and have a few recommendations (I can probably help implement it as well):
* The build system uses Makefiles, you really should switch to CMake. Having a look at the Makefile, you are not doing anything special so a conversion would go fast. Why you should go to CMake is the built in version handling, as you saw in Issue #69 this error would have been caught.
* The build is allowing warnings that really needs to be fixed (such as non handled cases in a switch), `-Werror` should be added to highlight these (and `-Wpedantic` to make sure that it's ISO C++ conforming as well, but that's not so important) 
* Adding a `.clang-format` to make it easier for contributors to add and keep a coherent style for the project.
* Add a check of formatting (based on above) to CI so PRs are automatically checked.
* Add a list of contributors, it helps in getting more people if their contributions are recognized.

Tell me if anything of these are things you want help with, I have some free time and have experience with all of it.

peterus, 2018-01-22:
Hi!

Totally agree to your points! I would like to add even some more:
* a linter is will find more "bugs" or potentiel issues in the code and should also be added.
* I also suggest to do a lot of code refactoring. I have found already some parts which can lead to issues.

regarding cmake:
I am working already on a switch to cmake: [my fork](https://github.com/peterus/horizon/tree/cmake_build)
the status is: working on linux, still have to test it on windows and some minor fixes to have it nicer.

korken89, 2018-01-22:
Cool!
I had a look at your CMakeLists.txt and it seems to have the most stuff, what I would add are the version requirements on the external libs.

On the rest of the stuff, it's minor and would go quite fast to add if it's desired.

carrotIndustries, 2018-01-22:
>  The build system uses Makefiles, you really should switch to CMake. Having a look at the Makefile, you are not doing anything special so a conversion would go fast. Why you should go to CMake is the built in version handling, as you saw in Issue #69 this error would have been caught.

It'd be great if you could coordinate efforts with @peterus so that we'll get one PR in the end.

> The build is allowing warnings that really needs to be fixed (such as non handled cases in a switch), -Werror should be added to highlight these (and -Wpedantic to make sure that it's ISO C++ conforming as well, but that's not so important)

That should be fairly easy for my code, but I'm not quite sure whether to patch included 3rd party code. While -Werror definitely helps with development, I don't think it's a good thing to have in the build system since it may prevent users with different compilers from building.

> - Adding a .clang-format to make it easier for contributors to add and keep a coherent style for the project.
> - Add a check of formatting (based on above) to CI so PRs are automatically checked.

Sounds like good idea, go ahead!

> Add a list of contributors, it helps in getting more people if their contributions are recognized.

Github provides https://github.com/carrotIndustries/horizon/graphs/contributors, but feel free to open a PR.

korken89, 2018-01-22:
I have a tentative `.clang-format` file as follows, it generates almost the same as what you have now.
The only feature I could not quite get is the indentation of access modifiers, so I made them a half indent.

`.clang-format`:
```
BasedOnStyle: LLVM
Language: Cpp
IndentWidth: 8
UseTab: Always
BreakBeforeBraces: Attach
AlwaysBreakBeforeMultilineStrings: true
AllowShortIfStatementsOnASingleLine: false
AllowShortLoopsOnASingleLine: false
AllowShortFunctionsOnASingleLine: false
IndentCaseLabels: false
AlignEscapedNewlinesLeft: false
AlignTrailingComments: true
AllowAllParametersOfDeclarationOnNextLine: false
AlignAfterOpenBracket: true
SpaceAfterCStyleCast: false
MaxEmptyLinesToKeep: 2
BreakBeforeBinaryOperators: NonAssignment
SortIncludes:    false
ContinuationIndentWidth: 8
NamespaceIndentation: All
AccessModifierOffset: -4
```

korken89, 2018-01-22:
I have added and tested the automatic style checking, and using your 8 wide tabs together with 80 columns has a few problems.
Would you be open to reducing it?

You can check the style here: https://github.com/korken89/horizon
And here you can check how CircleCI reports errors & success for style: https://circleci.com/gh/korken89/horizon/tree/master

I am getting close to the first PR for the styles.

peterus, 2018-01-22:
> He seems to have it covered on the CMake integration, but if @peterus wants help just tell me! :)

I would need some help in testing on windows. As I do not have a windows machine anymore, I would need to download a virtual machine to test it (and much more longer than on a native one). Can you help me here?
I still want to change the generation of the gitversion-file as the current implementation is not so nice.

> Agreed, it's for CI and development to catch bugs hiding in warnings and not letting it build up to a huge amount of warnings.

Maybe some checks should be just done when running on an PR. On development this is not fully needed.
As some people like it also under development (like me), I would make it possible with an cmake argument or something similar.

carrotIndustries, 2018-01-22:
Yeah, optionally specifying -Werror would be best. Personally, I'm not that fond of the 80 char line length limit since it leads to ugliness like this: https://github.com/korken89/horizon/blob/master/src/pool-mgr/pool_update_error_dialog.cpp#L5

Regarding indentation: My editor is set to to 4 wide tabs, so the access modifiers end up on the same level as the definitions. In general I don't think mixing tabs and spaces in that context is a good idea, so just put them at the same indent level as the rest of the class.

korken89, 2018-01-23:
Awesome, then it was my auto detect that flaked out! Would you be open to switching to spaces rather than tabs? This it a lot more portable across editors. Also, I can increase the columns if the file you referenced still looks bad after I update the style.

korken89, 2018-01-23:
@peterus haha, I am also only on a Linux system.. Though, you can test it through CI, no med for a VM

carrotIndustries, 2018-01-23:
I'm fine with spaces as well. In case there's something you can't test in CI, I got a Windows machine at hand.

korken89, 2018-01-23:
Perfect, I have updated the style. 
It looks a lot better now, if the current test succeeds I will open a PR for this.

CarloMara, 2018-01-26:
I too was porting the build system to Cmake so thanks @peterus for the great work.
Also @carrotIndustries do you have an eta for switching the main repo to Cmake? 

Thanks again to all,
Carlo

carrotIndustries, 2018-01-26:
I don't have an ETA, better ask @peterus when he'll open the PR for the switch to cmake.

peterus, 2018-01-26:
I still have to test it on Windows and have to adapt the build scripts for the CI.
I will try to find some time today in the evening to test the old branch, sync the latest changes from master and create a PR for you.

peterus, 2018-01-27:
Today i finally finished the cmake files for Windows after a long time playing around on my only Windows Laptop which is taking around 20 min to get all files compiled and than starting the linking. (it was soooo slow :( )
After this work i wanted to merge the changes from master to my branch. But there where so many changes that it was easier to get the latest master status and to apply my changes by hand again.
As this is taking again a lot of time i am planing to deliver the changes tomorrow.
Sorry for the delay!

korken89, 2018-01-29:
@peterus Awesome!
When the CMake additions are in I can add some flags for debug builds on CI to catch warning and all that, unless you have already added it?

peterus, 2018-01-31:
I am so sorry, but this was taking much longer than expected.
I also discovered two issues where I was spending already hours but could not resolve them. Maybe somebody else with cmake knowledge can take a look?

I created already an PR: #91

peterus, 2018-01-31:
@korken89 feel free to add the flags for more warnings after the changes are in main ;)

peterus, 2018-01-31:
What was just coming into my mind:
maybe we need some different c-flags on Windows for the resources?

peterus, 2018-02-12:
finally the changes for cmake are done in #111 
please review and try it for your own ;)

korken89, 2018-02-12:
Cool! I will give it a try, great work!