horizon-eda/horizon #76: Gtkmm 3 ver. 3.20 Requirement

fabriziotappero, 2018-01-25:
For a successful compilation, Horizon requires gtkmm-3 version >3.20. Because most of the current linux ditros (in my case Mint 18.3) come with gtk 3.19 this requirement is a bit of a problem. 

Is there an easy way to install gtkmm 3.22without a major Linux distro update?

carrotIndustries, 2018-01-25:
Gtk 3.19? That's a bit strange since the odd-numbered gtk versions are for development releases only. The regular versions are all even-numbered. The lowest required gtk version is 3.20.

> Is there an easy way to install gtkmm 3.22without a major Linux distro update?

Dunno, you're probably better off asking this question on a linux mint forum or so.

Anyhow, as I stated in https://github.com/carrotIndustries/horizon/pull/74#issuecomment-360290999 the upcoming ubuntu LTS will include a sufficiently recent gtk version, so my motivation of making horizon compatible with gtk < 3.20 is rather low.

If someone contributes a not overly messy, easily-revertible patch that adds a wrapper around GtkFileChooserNative, I'll very likely merge it.

fabriziotappero, 2018-01-26:
> Anyhow, as I stated in #74 (comment)

Hi Lukas,
thanks for the hint. Yes you are right. I meant version 3.18.

I compiled successfully horizon in a ubuntu 17 virtual environment but it
is too slow on my machine. I guess I'll wait for the next LTS to support
gtk >3.20.

cheers
Fabrizio

closed 2018-01-27 by carrotIndustries