horizon-eda/horizon #77: Color scheme situation

CarloMara, 2018-01-26:
Hi folks,

I got very excited when on the kicad mailing list horizon was posted, mainly because of the lack of legacy code. I have just one problem, the color scheme. I know that this may sound silly, but IMHO colors can be very helpful in understanding what you're looking at. 

The first problem that I see is granularity, or lack thereof. I'd like to be able to change the color of wires, without changing the one of busses. The option to change only the background it a very limiting factor. Kicad for example gives the option to customize mostly everything:
```
ColorWireEx=rgb(88, 110, 117)
ColorBusEx=rgb(38, 139, 210)
ColorConnEx=rgb(88, 110, 117)
ColorLLabelEx=rgb(88, 110, 117)
ColorHLabelEx=rgb(133, 153, 0)
ColorGLabelEx=rgb(159, 37, 35)
ColorPinNumEx=rgb(7, 54, 66)
ColorPinNameEx=rgb(7, 54, 66)
ColorFieldEx=rgb(101, 123, 131)
ColorReferenceEx=rgb(101, 123, 131)
ColorValueEx=rgb(101, 123, 131)
ColorNoteEx=rgb(38, 139, 210)
ColorBodyEx=rgb(0, 43, 54)
ColorBodyBgEx=rgb(238, 232, 213)
ColorNetNameEx=rgb(132, 132, 132)
ColorPinEx=rgb(0, 43, 54)
ColorSheetEx=rgb(132, 0, 132)
ColorSheetFileNameEx=rgb(181, 137, 0)
ColorSheetNameEx=rgb(0, 132, 132)
ColorSheetLabelEx=rgb(133, 153, 0)
ColorNoConnectEx=rgb(38, 139, 210)
ColorErcWEx=rgb(133, 153, 0)
ColorErcEEx=rgb(220, 50, 47)
ColorGridEx=rgb(171, 171, 171)
ColorBgCanvasEx=rgb(253, 246, 227)
```

The second problem that I see, it in the color selection. Contrast is important, but too much contrast tires the eye. I think that a different palette should be chosen. 

The third, and most silly of all problems, was pointed out by a friend: "We have altium licenses in the [uni] labs, why in hell are you using legacy software?" I know that this may sound stupid, and it really is, but the point risen was clear. A nice color scheme can be an important factor.

I want to be clear on this. I don't have the **best** answer. I want to start a discussion about this topic, because I think it's important. I don't know what a nice color scheme is, [solarized](http://ethanschoonover.com/solarized) or [base16](http://chriskempson.com/projects/base16/) could be starting points. I don't know if kicad fine tuning level is too much, but I'm definitely sure that a binary option is not good enough.
I'd really to get his feature done, but before doing anything I'd like get some feedback.

Cheers,
Carlo

P.S. I just read closed issue by @carrotIndustries regarding a proposal for a new logo. I am now seriously questioning weather or not the color scheme is like this on purpose. Anyways I still think that a discussion on the topic is in order.

carrotIndustries, 2018-01-26:
Yeah, the color scheme is on purpose, I've got a soft sport for that vector CRT look.
Anyhow, I received feedback from a colorblind user: http://www.eevblog.com/forum/eda/new-work-in-progress-eda-package!/msg1409946/#msg1409946, so customizable colors will definitely arrive in the near future.

CarloMara, 2018-01-26:
Cool, could you share some details on the implementation that you were thinking? I'd like to help if I can!

carrotIndustries, 2018-01-26:
First of all, we'll have to figure out in which places colors are defined. Right now, color definitions are scattered all over the place: Things on the canvas are colored using a palette https://github.com/carrotIndustries/horizon/blob/master/src/canvas/triangle.hpp#L9
These colors are set in https://github.com/carrotIndustries/horizon/blob/master/src/canvas/canvas.cpp#L12 and are pushed to the GPU as an UBO.
Layer colors are already user-settable (right click in the layer box) for "layered" canvases, i.e. padstack, package and board. For anything else (symbol, schematic), the `LayerProvider` default is used.

Other colors, such as the color of the selection boxes are hardcoded in the shaders: https://github.com/carrotIndustries/horizon/tree/master/src/canvas/shaders, We'll need to make these uniforms to be able to set these at run time.

To keep things simple, simply adding the new colors to the `ColorP` enum at an offset past `N_COLORS` should be a good way to go. The `Canvas` then keeps a map to store the value. For mapping the members of the `ColorP` enum to human-readable names, a map similar to `tool_catalog` seems reasonable.

Now onto where/how to store the colors: We already have the infrastructure for preferences, so let's use this one. I propose adding the color settings to the `CanvasPreferences` class since that way the user can set different colors for layered and non-layered canvases.

I hope that's enough to get you started. Don't hesitate to ask if there's anything that isn't yet clear.

CarloMara, 2018-01-27:
That's definitely a good starting point. I'm gonna wait until CMake support is merged because I'm more comfortable working with that instead.

On a side note would you mind elaborate the design idea that lead to the creation of layered and non layered canvas? I can open a new issue if you think that the topic is worth a more in depth discussion.

Thanks for the fast answer,
Carlo

carrotIndustries, 2018-01-27:
In terms of implementation layered and non-layered Canvas are both the same `Canvas` class. I created that distinction in the preferences so that we can have different visual settings for schematic and board for example.

atoav, 2018-01-30:
One thought from the graphics-/usability-side:  
It might be wise to implement a (optional) seperate template for printing. I typically don't mind using a black-and-white schematic (in Eagle I'd often use the "monochrome" switch), but I kinda like the idea to have a coloured export too.  

What is more important IMHO:
- To have the ability to safe and change these color patterns (you don't want to change 20 color variables by hand whenever you feel like change)
- Sensible defaults that kinda work for colorblind people  

From a UI standpoint it would make sense to group these settings together with other Theme-/Skin related Settings like Fontsizes, UI-Colorschemes etc. A typical approach would be the usage of JSON Theme-Files that can be loaded, exchanged and shared fast, but obviously I don't know how much would have to be touched in order to achieve that.

carrotIndustries, 2018-01-30:
>  I typically don't mind using a black-and-white schematic

That's how it's implemented right now. (Schematic export is on my list of things that a need major overhaul in the not too far future)

> A typical approach would be the usage of JSON Theme-Files that can be loaded, exchanged and shared fast, but obviously I don't know how much would have to be touched in order to achieve that.

Good news: That should come for next-to free since we'll need to save these as json for the preferences file anyhow.

CarloMara, 2018-03-12:
Hi everybody,

It took way more time than expected, but I'm back; This post is mainly questions  for @carrotIndustries, so feel free to skip it.

>Layer colors are already user-settable (right click in the layer box) for "layered" canvases, i.e. padstack, package and board. For anything else (symbol, schematic), the LayerProvider default is used.

I really don't get this. For example `void Canvas::render(const LineNet &line)` doesn't use the `LayerProvider` defaults, but actually specifies the color from the `ColorP` lists. What am I getting wrong? 

> Things on the canvas are colored using a palette https://github.com/carrotIndustries/horizon/blob/master/src/canvas/triangle.hpp#L9
These colors are set in https://github.com/carrotIndustries/horizon/blob/master/src/canvas/canvas.cpp#L12 and are pushed to the GPU as an UBO.

In order to better understand the code the first thing that I did was to change the values in `canvas.cpp`, but unfortunately that didn't affect symbol lines color. After a more careful look at the code, I think those are colored using the default form  `layer_provider.cpp`. Especially form the point risen the the aforementioned question, do you think it's a wise move to change to code in `render.cpp` to use the color definition form `canvas.cpp` ? 

I'm very sorry that this took so much time, but exams and stuff got in the way. I'm hoping to be more present in the near future,
Carlo

carrotIndustries, 2018-03-12:
> I really don't get this. For example void Canvas::render(const LineNet &line) doesn't use the LayerProvider defaults, but actually specifies the color from the ColorP lists. What am I getting wrong?

Perhaps my explanation was missing some detail. There's the special Color `ColorP::FROM_LAYER` that gets replaced by the respective layer's color (either default from `LayerProvider` or set by the user).

Lines can be used in layered (e.g. package) and non-layered (e.g. symbol) documents, so they'll need to be colored according to the layer they're on.

To sum it up: There are two ways of coloring things in horizon
 - Directly assign a color in the `Canvas::render` method. Mostly used for objects only present in non-layered canvases such as net lines and pins
 - Assign `ColorP::FROM_LAYER`and the object's color will be set by its layer. In layered canvases, the layer colors are settable by the user.

carrotIndustries, 2018-10-21:
32f94682bac38a510ffea64d405c8e7446c07faa makes all colors in the canvas user-settable. If people have some ideas for reasonable presets, I might make these available in the preferences dialog.

[image]

atoav, 2018-10-22:
Is there a AppVeyor build or should I build myself? I think having a preset selector makes sense in any way because people might change things and they maybe would like to reset back to the original.

carrotIndustries, 2018-10-22:
> Is there a AppVeyor build or should I build myself?

The latest appveyor build should work.

atoav, 2018-10-22:
Ok weird. I asked because I run the latest build and I don't see the color settings panes (everything else (Antialiasing yaddayadda looks the same)

atoav, 2018-10-23:
Okay, sorry. I had a mixup, it was a long day yesterday..  
Let's say I'd create a color scheme, is there a way to share it with you, so it can be included?

CarloMara, 2018-10-23:
It would be nice if color schemes could be committed as a part of libraries. It would make changing color scheme much more easier.

atoav, 2018-10-23:
Cool stuff.  
[image]

carrotIndustries, 2018-10-24:
5f4a233a410d761123496e15272bc82bfaabc9d5 adds importing/exporting colors. this will be the way of sharing color schemes. Presets should land within the next days.

carrotIndustries, 2018-10-30:
26f763b04ca2a01212fd66993b0264cbb82c3137 adds a layered flag to the saved color schemes to prevent loading the wrong one.

> I agree with @atoav that the current naming is a bit confusing, but I still see the fact the current naming is based on the code

I thought about naming it Schematic/Board display, but went with layer / non layer to express that it also affects symbol/padstack/package. I'm open to better names for these settings.

carrotIndustries, 2018-10-31:
ac40bc242ea319d2537de8214f21db24857d92f3 renames the settings to "Schematic/Board appearance"

carrotIndustries, 2018-11-01:
>  Let's say I'd create a color scheme, is there a way to share it with you, so it can be included?

Export the color scheme and attach it to this issue. After I've included it, it'll show up in load/save menu as well.

atoav, 2018-11-01:
## Boring (Schematic)
[image]
## Nostalghia (Schematic)
[image]

Both to be found here in a zip file (github won't allow .json):  
[horizon_color_schemes.zip](https://github.com/carrotIndustries/horizon/files/2540376/horizon_color_schemes.zip)

carrotIndustries, 2018-11-02:
c8896047be8fd1ba5565b6f7c7a6c62bd639a088 adds these presets
add new presets here: https://github.com/carrotIndustries/horizon/tree/master/src/preferences/color_presets

carrotIndustries, 2018-12-09:
> I think having some sort of standard preview file that displays all possible things (buses, nets, multiple layers etc.) would really help. Do you (or anybody) happen to have something lying around that fits the bill?

https://github.com/carrotIndustries/x-band-tx should have instances of every item supported in horizon.

carrotIndustries, 2019-02-06:
Closing since colors are now configurable by the user.

closed 2019-02-06 by carrotIndustries