horizon-eda/horizon #78: horizon-prj: crash at launch

jeanthom, 2018-01-30:
I'm running Fedora x86_64. I compiled horizon using the project's Makefile. When launching horizon-prj in a terminal, I get this :

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
fish: './horizon-prj' terminated by signal SIGABRT (Abandon)
```

Here is the backtrace from GDB :

```
#0  0x00007ffff56f28df in raise () at /lib64/libc.so.6
#1  0x00007ffff56f44da in abort () at /lib64/libc.so.6
#2  0x00007ffff60344fd in __gnu_cxx::__verbose_terminate_handler() () at /lib64/libstdc++.so.6
#3  0x00007ffff60322b6 in  () at /lib64/libstdc++.so.6
#4  0x00007ffff6032301 in  () at /lib64/libstdc++.so.6
#5  0x00007ffff6032519 in  () at /lib64/libstdc++.so.6
#6  0x00007ffff605c357 in std::__throw_out_of_range_fmt(char const*, ...) () at /lib64/libstdc++.so.6
#7  0x000000000041530c in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::_M_range_check(unsigned long) const (__n=1, this=0x7fffffffd700) at /usr/include/c++/6.4.1/bits/stl_vector.h:804
#8  0x000000000041530c in std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::at(unsigned long) (__n=1, this=0x7fffffffd700) at /usr/include/c++/6.4.1/bits/stl_vector.h:825
#9  0x000000000041530c in main(int, char**) (c_argc=<optimized out>, c_argv=<optimized out>) at src/prj-util/util_main.cpp:45
```

jeanthom, 2018-01-30:
Seems related to a missing argc check :

```c
    if (argv.at(1) == "create-block") {
        horizon::Block block(horizon::UUID::random());
        if (argv.size() >= 4) {
            block.name = argv.at(3);
        }
        save_json_to_file(argv.at(2), block.serialize());
    }
    else if (argv.at(1) == "create-schematic") {
        horizon::Pool pool(pool_base_path);
        auto block = horizon::Block::new_from_file(argv.at(3), pool);
        horizon::Schematic sch(horizon::UUID::random(), block);
        save_json_to_file(argv.at(2), sch.serialize());
    }
    else if (argv.at(1) == "create-board") {
        horizon::Pool pool(pool_base_path);
        auto block = horizon::Block::new_from_file(argv.at(3), pool);
        horizon::Board brd(horizon::UUID::random(), block);
        save_json_to_file(argv.at(2), brd.serialize());
    }
```

carrotIndustries, 2018-01-30:
Both binaries aren't exactly meant for public consumption anymore since we now have the pool and project manager (`horizon-pool-mgr` and `horizon-prj-mgr`). Documentation on both binaries is there: https://github.com/carrotIndustries/horizon/wiki/CLI-usage

But yeah, the argv parsing could really be improved on these and other binaries.

closed 2018-02-06 by carrotIndustries