horizon-eda/horizon #79: Package needed to build on Fedora

jeanthom, 2018-01-30:
Hi !

I'm trying to list the required Fedora packages so that other Fedora users could build Horizon easily. So far I listed those packages :

 * cppzmq-devel
 * OCE-devel
 * gtkmm30-devel
 * libgit2-devel
 * libuuid-devel
 * yaml-cpp-devel
 * sqlite-devel
 * librsvg2-devel
 * cairomm-devel
 * glm-devel
 * boost-devel
 * curl

If someone could try to reproduce this on a clean Fedora install, that would be great ! :smiley:

carrotIndustries, 2018-02-06:
Can you provide something that people can paste into their shell similar to: https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Linux?

jeanthom, 2018-02-06:
I will do a clean Fedora 27 install to check which dependencies are needed on a clean Fedora system (incl. all build tools like gcc).

jeanthom, 2018-02-06:
Build is failing due to incompatibility between GCC 7.3.0 and current version of GLM. I just sent an email to glm-devel maintainer so that he/she could fix that.

So far I've used those packages :
```
sudo dnf install git make gcc gcc-c++ pkg-config cppzmq-devel OCE-devel gtkmm30-devel libgit2-devel libuuid-devel yaml-cpp-devel sqlite-devel librsvg2-devel cairomm-devel glm-devel boost-devel libcurl-devel
```

jeanthom, 2018-02-06:
Got in touch with the maintainer, seems like GCC 7.2 should be the default compiler on Fedora 27. I can get a working build with those packages :

```
sudo dnf install git make gcc-7.2.1-2.fc27 gcc-c++-7.2.1-2.fc27 pkg-config cppzmq-devel OCE-devel gtkmm30-devel libgit2-devel libuuid-devel yaml-cpp-devel sqlite-devel librsvg2-devel cairomm-devel glm-devel boost-devel libcurl-devel
```

Could you add that into the wiki ?

carrotIndustries, 2018-02-07:
Done

closed 2018-02-07 by carrotIndustries

jeanthom, 2018-02-07:
https://bugzilla.redhat.com/show_bug.cgi?id=1542941