horizon-eda/horizon #82: Pasting Text doesn't work

atoav, 2018-01-30:
I can't paste text into any text field on Windows 10 x64. This is annoying because I am too lazy to enter my 64 character password manually, to contribute to the pool.

jeanthom, 2018-01-31:
I just tested this on Fedora, works ok. Seems like a Windows-only issue.

carrotIndustries, 2018-01-31:
I tested this on my Windows 10 machine, both Ctrl-V and context menu paste work.

atoav, 2018-01-31:
Okay, I tried to replicate today (yesterday was past midnight already):
1. I run`D:\horizon\horizon-prj-mgr.exe`
2. Pasting into the  Field `Project Title` works (this example used the string "Foo")
3. Close it

Then:
1. I run `D:\horizon\horizon-pool-mgr.exe`
2. I go to the `Remote`-Tab
3. Pasting worked (again the string "Foo")
4. I go to `Parts`-Tab and click `Create`
5. I try to copy and paste text from Firefox or Notepad in any field (another text "Bar")
6. "Foo" gets pasted and not "Bar"

carrotIndustries, 2018-01-31:
Congrats, you just found a new Gtk bug! I created ticket in the Gtk bugtracker: https://bugzilla.gnome.org/show_bug.cgi?id=793078 Maybe I'll find some time to look into this myself...

carrotIndustries, 2018-02-26:
Fixed in gtk 3.22.27/28, should be fixed in latest appveyor build or

closed 2018-02-26 by carrotIndustries