horizon-eda/horizon #85: Pool-Manager: Remote login fails with active two factor authentification

atoav, 2018-01-31:
If you have two-factor authentification enabled in Github, the login fails without displaying a helpful error message. Maybe in the long term there needs to be a solution for this. A quick fix would be to display some advice about this.

jeanthom, 2018-02-06:
Can you try this procedure ?
https://help.github.com/articles/creating-a-personal-access-token-for-the-command-line/

RX14, 2020-09-10:
This procedure works, but is deprecated (github sends you an email to stop logging in like this). Also, credentials are not saved in horizon eda, so you have to save a personal access token manually somewhere.

carrotIndustries, 2020-10-05:
This has been fixed by 69c8b92a455e494179ea76e86d2de5d1458ef97b.

closed 2020-10-05 by carrotIndustries