horizon-eda/horizon #86: "To board" button in schematic editor does not work

jeanthom, 2018-01-31:
The "To board" button in the schematic editor does not launch board editor. It only produces the following message in console :
```
imp rx null
```

Platform: Fedora x86_64

carrotIndustries, 2018-01-31:
Perhaps, this button is a bit mislabeled: If a board editor is running, it instructs the board editor to start the 'place package' tool with the symbols selected in the schematic.
The optimum behavior would be to launch the board editor and start the tool as if it was running. Unfortunately, this is bit difficult due to the async nature of things.

To make this work, these things are necessary: Project manager receives "to board" message, launches board editor, board editor calls back to the project manager "I'm ready", project manager delivers the "to board" message to the board editor.

Worth the complexity? Or just pop up a dialog "sorry, can't do that"?

jeanthom, 2018-01-31:
I was expecting something more like what a similar button in Eagle & KiCAD does : launching board editor

Maybe the "To board" button is not worth a constant display and should be a menu option when you right click on a component (like "place component in board view") ?

atoav, 2018-01-31:
I was also mislead by that button and until now thought it was just broken I agree with @jeanthom: the text should be changed and maybe it should be placed into the Spacebar-Menu with a nice shortcut.

jeanthom, 2018-02-01:
Some quick notes for later work :
 * "To board" button is defined here :
https://github.com/carrotIndustries/horizon/blob/7948516ad84c403e27d0650ae70398eabe77fe64/src/imp/imp_schematic.cpp#L296-L311
 * Tools are defined here :
https://github.com/carrotIndustries/horizon/blob/6b7ca4c0a2726a03523a4f33b2af4087ac2b5844/src/core/tool_catalog.cpp
https://github.com/carrotIndustries/horizon/blob/7948516ad84c403e27d0650ae70398eabe77fe64/src/core/core.hpp
I'm not sure where are the one related to schematic view, maybe here ?
https://github.com/carrotIndustries/horizon/blob/7948516ad84c403e27d0650ae70398eabe77fe64/src/imp/imp_schematic.cpp

 * Board editor is opened in project manager using this code :
https://github.com/carrotIndustries/horizon/blob/6b7ca4c0a2726a03523a4f33b2af4087ac2b5844/src/prj-mgr/prj-mgr-app_win.cpp#L171-L180

carrotIndustries, 2018-02-01:
I won't recommend tackling this issue until #92 is resolved since there will be a "to board" action afterwards.

commit 32f4b57e (2018-02-25): make "to board" an action

also fix #86
closed 2018-02-25 by carrotIndustries