horizon-eda/horizon #90: Pool-Manager

atoav, 2018-01-31:
I made footprints for DIP-packages. Whenever I try to create a pull requests I end up with a: `Error: timeout waiting for fork` after a few minutes. It is unclear to me how to pinpoint the reason, I selected Packages to merge, added a commit title and a pull request message. 

I could of course go the manual/CLI-route with git, but I think the ability to contribute to the pool in a easy way is important for the project to get the traction it deserves.

jeanthom, 2018-01-31:
Have you tried with an increased value for retries variable ? Like 1000 for instance.

https://github.com/carrotIndustries/horizon/blob/124aad444732436ed0a13f3bdd0259747f88f240/src/pool-mgr/pool_remote_box.cpp#L772-L791

atoav, 2018-02-01:
I didn't compile it myself but I will try to and report back once I know if that helps.
My C++ skills are not where they should be (more of a python guy).

atoav, 2018-02-01:
Okay I managed to get a build with a timeout of 600 seconds or 10 minutes and still got a `Error: timeout waiting for fork` after that period – so there might be some other problem stopping me from getting a fork. I have got quite a cluttered system right now and planned to do a fresh OS-Install within the next month, so maybe these problems will solve themselves.

Is there any guide how to properly contribute to the pool manually via CLI?

carrotIndustries, 2018-02-01:
From what I can tell, the fork went well: https://github.com/atoav/horizon-pool. You may try deleting the repo and fork again.

atoav, 2018-02-01:
I did the following:
1. Delete the forked repo `horizon-pool` via githubs settings
2. upgraded the pool
3. selected all items to merge in the Tab `Packages`
4. entered a commit title and pull request message
5. press `Create pull request` and enter credentials
6. wait... 
7. `Error: timeout waiting for fork`

In `horizon-pool/packages/ic/` on the forked repo there is only a `smd` directory I created a `th` directory within, this is not visible in github. So obviously I forked, but after that nothing happens for some reason.

carrotIndustries, 2018-02-06:
Try 113d15b31b8835ed40cd636595fc1720b24bbe1b, what gets displayed after "Waiting for fork"?

atoav, 2018-02-10:
Delayed answer: I currently have nothing to fork, and it took some time to get my system back up to speed. I will test as soon as I have something that I can fork

atoav, 2018-02-11:
Okay I did a git clone and a fresh compile from source with 113d15b31b8835ed40cd636595fc1720b24bbe1b included. Ilogged in with my regular github mail and password (I wanted to try the token I made for this but #82 is giving me hard time entering a random 64 char password by hand – and if it doens't work I don't know wether it was a typo or a bug).    

Long story short: I still get a `Error: timeout waiting for fork`

carrotIndustries, 2018-02-11:
Okay, but what does it show while waiting for fork?

atoav, 2018-02-11:
Ah right now I get a `Waiting for fork... (unexpected HTTP response code 404 when accessing https://api.github.com/repos/dh@atoav.com/horizon-pool)` with a rotating wheel besides it

atoav, 2018-02-11:
Great – this works now! So the user needs to know that they have to specify their github username not the email they use to log into github : )

atoav, 2018-02-20:
So like it is right now is still somehow broken behaviour, because it lets you login with the email adress, but fails at the fork because it is deriving the URL to request from the entered user name.

From a UI/UX position I suggest one of the following solutions to avoid people complaining about not beeing able to fork, like I did:
- have a placeholder value that shows what kind of username is expected (eg. `carrotindustries`) to avoid ambiguity
- derive the user's username from their email adress if they entered one and use that for the fork request URL

For people that use a password manager (like I do) the second option would be more appealing because password managers typically store email and password next to each other and don't really care about a username.  

In the long term we should probably think about a solution based on github tokens, were people don't have to enter their credentials all the time and where they don't have to trust horizon not to do weird things on their other (valuable) repositories, because they can limit what horizon is allowed to do with their account.

carrotIndustries, 2018-02-20:
> derive the user's username from their email adress if they entered one and use that for the fork request URL

That was surprisingly easy: 1f4c33d8c80f336903c7bd64c4f143935c9d6fb8 

> In the long term we should probably think about a solution based on github tokens

Tokens should already work since these can be used in place of a password.

atoav, 2018-02-20:
Great! so I think we can close this!

closed 2018-02-20 by atoav