horizon-eda/horizon #93: Failing build on Ubuntu 16.04

spiliot, 2018-02-03:
Using make after installing the packages mentioned in the wiki breaks with
```
src/imp/imp_schematic.cpp: In member function ‘void horizon::ImpSchematic::handle_export_pdf()’:
src/imp/imp_schematic.cpp:334:3: error: ‘GtkFileChooserNative’ was not declared in this scope
   GtkFileChooserNative *native = gtk_file_chooser_native_new ("Save PDF",
   ^
src/imp/imp_schematic.cpp:334:25: error: ‘native’ was not declared in this scope
   GtkFileChooserNative *native = gtk_file_chooser_native_new ("Save PDF",
                         ^
src/imp/imp_schematic.cpp:338:13: error: ‘gtk_file_chooser_native_new’ was not declared in this scope
    "_Cancel");
             ^
src/imp/imp_schematic.cpp:348:54: error: ‘GTK_NATIVE_DIALOG’ was not declared in this scope
   if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                      ^
src/imp/imp_schematic.cpp:348:55: error: ‘gtk_native_dialog_run’ was not declared in this scope
   if(gtk_native_dialog_run (GTK_NATIVE_DIALOG (native))==GTK_RESPONSE_ACCEPT) {
                                                       ^
Makefile:499: recipe for target 'src/imp/imp_schematic.o' failed
```
Is 17.10 necessary? Any workarounds? 

Also, would creating a containerized application with CI be possible for users that are not knowledgeable enough or don't have a proper build system?

jeanthom, 2018-02-03:
GtkFileChooserNative is part of Gnome >= 3.20. In Ubuntu Xenial/16.04, you have Gnome 3.18 at best. Two options : upgrade to a newer version of Ubuntu, or modify the code to use another GtkFileChooser.

Related : https://github.com/her001/Asma/issues/14

spiliot, 2018-02-04:
It looks like this was the case indeed, thanks for the pointer. Since I'm not at all familiar with the intricacies of changing the file picker I opted to update the gnome version in my system, it seemed relatively risk free. 18.04LTS is a couple of months away where that will be "fixed".

Now build is failing for another reason:
```
In file included from src/imp/imp_board.cpp:6:0:
src/imp/step_export_window.hpp:29:10: error: ‘mutex’ in namespace ‘std’ does not name a type
     std::mutex msg_queue_mutex;
          ^
```
Any ideas?

carrotIndustries, 2018-02-05:
Try adding `#include <mutex>`  to `src/imp/step_export_window.hpp` Does this fix things?

spiliot, 2018-02-05:
It did and thanks a lot. Now I hit another one 🙁

```
src/dialogs/dialogs.cpp: In member function ‘std::tuple<bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double> horizon::Dialogs::ask_dxf_filename(horizon::Core*)’:
src/dialogs/dialogs.cpp:430:52: error: converting to ‘std::tuple<bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double>’ from initializer list would use explicit constructor ‘constexpr std::tuple< <template-parameter-1-1> >::tuple(_UElements&& ...) [with _UElements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, long int&, double&}; <template-parameter-2-2> = void; _Elements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double}]’
         return {true, filename, layer, width, scale};
                                                    ^
src/dialogs/dialogs.cpp:433:53: error: converting to ‘std::tuple<bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double>’ from initializer list would use explicit constructor ‘constexpr std::tuple< <template-parameter-1-1> >::tuple(_UElements&& ...) [with _UElements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, int&, long int&, double&}; <template-parameter-2-2> = void; _Elements = {bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long int, double}]’
         return {false, filename, layer, width, scale};
                                                     ^
```
I wish I could fix things myself but when it comes to C++ I understand practically nothing.

jeanthom, 2018-02-05:
I am currently attempting a build on a fresh Ubuntu 17.10 install. We will see if I can manage to reproduce your build issue.

carrotIndustries, 2018-02-05:
17.10 should work fine since the CI uses that version of ubuntu.

> It did and thanks a lot. Now I hit another one 🙁
That's the compiler being not as smart as I was hoping for...

Try updating gcc as well.

jeanthom, 2018-02-05:
horizon is built without any problem. I used the following Ubuntu packages :
 * git
 * libxm2-util
 * build-essential
 * pkg-config
 * libglib2.0-dev
 * libsqlite3-dev
 * libyaml-cpp-dev
 * libgtkmm-3.0-dev
 * librsvg2-devel
 * libzmqpp-dev
 * libgit2-dev
 * uuid-dev
 * liboce-ocaf-dev
 * libglm-dev

spiliot, 2018-02-08:
Originally 16.04 comes with gcc-5, I was able to update it to 6 but 7 is impossible as package requirements can not be met.

With 6 I get passed the error mentioned before and hit:
```
src/property_panels/property_panel.cpp: In static member function ‘static horizon::PropertyPanel* horizon::PropertyPanel::create(horizon::ObjectType, horizon::Core*, horizon::PropertyPanels*)’:
src/property_panels/property_panel.cpp:122:51: error: no matching function for call to ‘Gtk::Builder::get_widget_derived(const char [14], horizon::PropertyPanel*&, horizon::ObjectType&, horizon::Core*&)’
     x->get_widget_derived("PropertyPanel", w, t, c);
                                                   ^
In file included from /usr/include/gtkmm-3.0/gtkmm.h:119:0,
                 from src/property_panels/property_panel.hpp:2,
                 from src/property_panels/property_panel.cpp:1:
/usr/include/gtkmm-3.0/gtkmm/builder.h:726:8: note: candidate: template<class T_Widget> void Gtk::Builder::get_widget_derived(const Glib::ustring&, T_Widget*&)
   void get_widget_derived(const Glib::ustring& name, T_Widget*& widget)
        ^~~~~~~~~~~~~~~~~~
/usr/include/gtkmm-3.0/gtkmm/builder.h:726:8: note:   template argument deduction/substitution failed:
src/property_panels/property_panel.cpp:122:51: note:   candidate expects 2 arguments, 4 provided
     x->get_widget_derived("PropertyPanel", w, t, c);
                                                   ^
In file included from /usr/include/gtkmm-3.0/gtkmm.h:119:0,
                 from src/property_panels/property_panel.hpp:2,
                 from src/property_panels/property_panel.cpp:1:
/usr/include/gtkmm-3.0/gtkmm/builder.h:771:8: note: candidate: template<class T_Widget> void Gtk::Builder::get_widget_derived(const Glib::ustring&, const T_Widget*&) const
   void get_widget_derived(const Glib::ustring& name, const T_Widget*& widget) const
        ^~~~~~~~~~~~~~~~~~
/usr/include/gtkmm-3.0/gtkmm/builder.h:771:8: note:   template argument deduction/substitution failed:
src/property_panels/property_panel.cpp:122:51: note:   candidate expects 2 arguments, 4 provided
     x->get_widget_derived("PropertyPanel", w, t, c);
                                                   ^
Makefile:501: recipe for target 'src/property_panels/property_panel.o' failed
```
Doesn't look like a compiler error anymore, probably another older package?

jeanthom, 2018-02-20:
To me it looks like a compiler issue... Ubuntu 17.10+ seems like a requirement

carrotIndustries, 2018-02-20:
As Kliment wrote on IRC, this is due to gtkmm <3.20 not supporting passing additional arguments with `get_widget_derived`.

jeanthom, 2018-02-20:
So we should close this issue?

spiliot, 2018-02-21:
As far as I'm concerned and assuming there's no quick workaround, yes. Thanks everyone for helping.

carrotIndustries, 2018-02-21:
Unless someone contributes an easily-revertible patch to support ubuntu 16.04 it will stay unsupported.

closed 2018-02-21 by carrotIndustries