horizon-eda/horizon #94: Difficult to contact program author

wookey, 2018-02-04:
There is no email address in the horizon source that I could find (nor indeed does the Author's name appear).
This makes it difficult to contact him (if for example you were trying to package his software). There is also no email in the FOSDEM slides. 'Lukas Kramer' online seems mostly to be a male model in Malaysia. Github doesn't appear to have a 'PM' facility. So I am resorting to a github issue.
Please tell wookey@wookware.org how to get in touch. I have a half-arsed proof-of-concept package already, and a few queries.

jeanthom, 2018-02-04:
I agreeing that his email address isn't easily available, but you can pull the git repository and then find his email address from git log.

Otherwise for project-related things, please use GitHub's issues & pull requests so that all the people currently working on this project could interact with you and the suggest.

rnestler, 2018-02-04:
It would also be nice to have an IRC or gitter or whatever channel available to discuss and get in touch with Lucas!

CarloMara, 2018-02-04:
Yhea, the irc would be nice

wookey, 2018-02-04:
On 2018-02-04 15:29 +0000, Carlo Maragno wrote:
> Yhea, the irc would be nice

I opened a #horizon-eda channel on freenode. Feel free to join me :-)
There is already a #horizon channel, although it is empty as it is the
name of some previous project. We could take that over if anyopne
cares about the name.

And yes - good point about the git comit info. I was too dumb to think of that :-)

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

peterus, 2018-02-05:
why not using something like "Hipchat"?
In hipchat the messages are stored and we can comment also later on some textes. it has also github, CI, etc. integration which is informing the participants and you can comment on new builds or other stuff.

carrotIndustries, 2018-02-05:
IRC is the way to go. I don't like these newfangled siloed messaging apps.

commit 28b5f76e (2018-02-05): add IRC channel to README

fixes #94
closed 2018-02-05 by carrotIndustries