horizon-eda/horizon #97: Continuous integration on Linux platforms

jeanthom, 2018-02-06:
Currently CI is only done on Windows (with Appveyor). I think we should have CI also for Linux targets. 

I don't know much about the free CI solutions on market, the only one I used was Travis CI (on my [image] project) which unfortunately does not provides support for recent Ubuntu versions.

Maybe something Docker based or a self-hosted Jenkins could be a solution.

carrotIndustries, 2018-02-06:
We do have circleci for linux: https://circleci.com/gh/carrotIndustries/horizon It's only used for verifying it builds though.

jeanthom, 2018-02-06:
Uh, didn't know that !

Currently only Appveyor build status is displayed whenever I do a pull-request. Is it possible to get both Appveyor and Circle CI integration in GitHub ?

closed 2018-02-06 by jeanthom

carrotIndustries, 2018-02-06:
Seems like building PRs is off by default on circleci: https://circleci.com/docs/1.0/fork-pr-builds/ I enabled that setting, let's see if it actually works.

jeanthom, 2018-02-06:
https://github.com/carrotIndustries/horizon/pull/98
Testing it right now.

jeanthom, 2018-02-06:
Works ok !