horizon-eda/horizon #100: Precompiled headers

jeanthom, 2018-02-07:
Build time is currently huge. On a dual-core multi-threaded CPU, it is around 11 minutes. We may be able to cut that time using precompiled headers :
 * https://gcc.gnu.org/onlinedocs/gcc/Precompiled-Headers.html
 * http://itscompiling.eu/2017/01/12/precompiled-headers-cpp-compilation/

jeanthom, 2018-02-07:
https://github.com/jeanthom/horizon/tree/precompiled-headers

I got compile time to around 8-9 minutes. However I don't like the principle of precompiled headers for a whole project.

carrotIndustries, 2018-02-23:
>  However I don't like the principle of precompiled headers for a whole project.

Me neither. Closing since probably not worth the effort.

closed 2018-02-23 by carrotIndustries