horizon-eda/horizon #106: Failing to build on Manjaro Linux 17.1

adrian1602, 2018-02-08:
Using make after installing all recommended packages for Arch Linux failes with following error:

`g++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -fdata-sections -ffunction-sections -pthread -I/usr/include/uuid -I/usr/include/gtkmm-3.0 -I/usr/lib/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/pool/unit.cpp -o src/pool/unit.o
In file included from /usr/include/yaml-cpp/node/node.h:17:0,
                 from /usr/include/yaml-cpp/yaml.h:16,
                 from src/pool/unit.hpp:9,
                 from src/pool/unit.cpp:1:
/usr/include/yaml-cpp/node/ptr.h:11:10: schwerwiegender Fehler: boost/shared_ptr.hpp: Datei oder Verzeichnis nicht gefunden
 #include <boost/shared_ptr.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~
Kompilierung beendet.
make: *** [Makefile:501: src/pool/unit.o] Fehler 1
`

Are there some packages missing in my System? Where can i get them?

carrotIndustries, 2018-02-08:
Oh, looks like I forgot to include the `boost` dependency.

adrian1602, 2018-02-09:
Now this error occured:

`src/util/step_importer.cpp:43:10: schwerwiegender Fehler: glm/glm.hpp: Datei oder Verzeichnis nicht gefunden
 #include <glm/glm.hpp>
          ^~~~~~~~~~~~~
Kompilierung beendet.
make: *** [Makefile:507: src/util/step_importer.o] Fehler 1
`

jeanthom, 2018-02-09:
I think you're missing the glm package : https://www.archlinux.org/packages/community/any/glm/

adrian1602, 2018-02-10:
It works thanks. You should ad the glm package to the dependecies

carrotIndustries, 2018-02-10:
Fixed it

closed 2018-02-10 by carrotIndustries