horizon-eda/horizon #108: Schematic editor : net name field has weird glitches

jeanthom, 2018-02-08:
[image]

Text orientation is right-to-left, and sometimes becomes a password field.

carrotIndustries, 2018-02-09:
The entries as well as most of the other editors are right-aligned to separate their content from the header above.

The dots went there to provide some sort of "editing in progress" feedback. No idea if that's still required.

jeanthom, 2018-02-21:
I think the dots are confusing as a progress indicator, a [Gtk::Spinner](https://developer.gnome.org/gtkmm/stable/classGtk_1_1Spinner.html) seems more appropriate

commit b72814ab (2018-02-23): remove confusing editing in progess indicator

fixes #108
closed 2018-02-23 by carrotIndustries