horizon-eda/horizon #110: Building with GCC 7.3.0 on Fedora

jeanthom, 2018-02-11:
Currently Fedora package list specifies an older GCC version (7.2) than the current GCC version (7.3), because otherwise the GLM library won't compile.

This issue is being sorted out, patches for GLM will arrive soon : https://bugzilla.redhat.com/show_bug.cgi?id=1542941

Once GLM is updated, we'll need to update the package list for Fedora to remove the GCC version requirement.

jeanthom, 2018-02-13:
Patches are now in Fedora 26 & 27 stable.

@carrotIndustries could you replace :

> On Fedora 27:
>
> ```
> sudo dnf install git make gcc-7.2.1-2.fc27 gcc-c++-7.2.1-2.fc27 pkg-config cppzmq-devel OCE-devel\
>    gtkmm30-devel libgit2-devel libuuid-devel yaml-cpp-devel sqlite-devel librsvg2-devel\
>    cairomm-devel glm-devel boost-devel libcurl-devel
> ```

with

> On Fedora 25/26/27:
>
> ```
> sudo dnf install git make gcc gcc-c++ pkg-config cppzmq-devel OCE-devel\
>    gtkmm30-devel libgit2-devel libuuid-devel yaml-cpp-devel sqlite-devel librsvg2-devel\
>    cairomm-devel glm-devel boost-devel libcurl-devel
> ```

carrotIndustries, 2018-02-13:
done

closed 2018-02-13 by carrotIndustries