horizon-eda/horizon #112: Pool-Manager: Download pool crash

atoav, 2018-02-12:
I managed to produced this wonderful error:  
[image]  

What I did:  
1. Download the Windows Build for 4e0e77d3fe80e6748958a6699398e3f02e475680 from [AppVeyor](https://ci.appveyor.com/project/carrotIndustries/horizon/build/artifacts)
2. Move the folder `horizon` to disk and start `horizon-pool-mgr.exe`
3. Click on `Download...`
4. Select Destination Directory – Other (A empty folder on my drive, let's call it `pool`)
5. Click the blue button `Download`
6. It crashes while updating the remote pool (pictured above)  

In the folder `pool` I find the following (output of `tree`):  

```
pool:.                                          
└───.remote                                  
    ├───3d_models                            
    │   ├───ic                               
    │   │   ├───qfn                          
    │   │   └───qfp                          
    │   └───passive                          
    │       ├───capacitor                    
    │       └───resistor                     
    ├───entities                             
    │   ├───connectors                       
    │   │   ├───generic                      
    │   │   ├───rj45                         
    │   │   └───usb                          
    │   ├───diode                            
    │   ├───ic                               
    │   │   ├───adc                          
    │   │   │   └───ltc                      
    │   │   ├───mcu                          
    │   │   │   ├───microchip                
    │   │   │   └───stm                      
    │   │   ├───regulator                    
    │   │   ├───timer                        
    │   │   └───transceiver                  
    │   │       └───ethernet                 
    │   ├───logic                            
    │   ├───opto                             
    │   ├───passive                          
    │   │   ├───capacitors                   
    │   │   ├───magnetics                    
    │   │   └───resistors                    
    │   └───transistors                      
    ├───packages                             
    │   ├───connectors                       
    │   │   ├───rj45                         
    │   │   │   └───RJHSE-5381               
    │   │   └───usb                          
    │   │       └───10118194-0001LF          
    │   │           └───padstacks            
    │   ├───ic                               
    │   │   └───smd                          
    │   │       ├───qfp                      
    │   │       │   └───lqfp-64              
    │   │       └───soic                     
    │   │           ├───so-14                
    │   │           ├───so-20                
    │   │           └───so-8                 
    │   ├───manufacturer                     
    │   │   ├───microchip                    
    │   │   │   ├───sqfn32                   
    │   │   │   └───tqfp176                  
    │   │   ├───stm                          
    │   │   │   └───ufqfpn20                 
    │   │   │       └───padstacks            
    │   │   └───ti                           
    │   │       ├───dcy                      
    │   │       └───rgc                      
    │   │           └───padstacks            
    │   └───passive                          
    │       ├───magnetics                    
    │       │   └───S558-5500-25-F           
    │       └───smd                          
    │           ├───C0402                    
    │           ├───C0603                    
    │           ├───C0603-manual_soldering   
    │           ├───C0805                    
    │           ├───C1206                    
    │           ├───R0402                    
    │           ├───R0603                    
    │           ├───R0603-manual_soldering   
    │           ├───R0805                    
    │           └───R1206                    
    ├───padstacks                            
    ├───parts                                
    │   ├───connector                        
    │   │   ├───rj45                         
    │   │   └───usb                          
    │   ├───ic                               
    │   │   ├───adc                          
    │   │   │   └───ltc                      
    │   │   ├───logic                        
    │   │   │   └───ti                       
    │   │   ├───mcu                          
    │   │   │   ├───microchip                
    │   │   │   └───stm                      
    │   │   ├───regulator                    
    │   │   │   └───ti                       
    │   │   ├───timer                        
    │   │   │   └───st                       
    │   │   └───transceiver                  
    │   │       └───ethernet                 
    │   └───passive                          
    │       ├───capacitors                   
    │       │   └───cpl                      
    │       ├───magnetics                    
    │       │   └───ferrite_beads            
    │       └───resistors                    
    │           ├───cpl                      
    │           └───yago                     
    ├───symbols                              
    │   ├───common                           
    │   ├───connectors                       
    │   │   ├───generic                      
    │   │   ├───rj45                         
    │   │   └───usb                          
    │   ├───diodes                           
    │   ├───ic                               
    │   │   ├───adc                          
    │   │   │   └───ltc                      
    │   │   ├───mcu                          
    │   │   │   ├───microchip                
    │   │   │   └───stm                      
    │   │   ├───regulator                    
    │   │   ├───timer                        
    │   │   └───transceiver                  
    │   │       └───ethernet                 
    │   ├───logic                            
    │   ├───opto                             
    │   │   └───leds                         
    │   ├───passive                          
    │   │   ├───capacitors                   
    │   │   ├───magnetics                    
    │   │   └───resistors                    
    │   └───transistors                      
    ├───tmp                                  
    └───units                                
        ├───common                           
        ├───connectors                       
        │   ├───generic                      
        │   ├───rj45                         
        │   └───usb                          
        ├───diodes                           
        ├───ic                               
        │   ├───adc                          
        │   │   └───ltc                      
        │   ├───mcu                          
        │   │   ├───microchip                
        │   │   └───stm                      
        │   ├───regulators                   
        │   ├───timer                        
        │   └───transceiver                  
        │       └───ethernet                 
        ├───logic                            
        ├───opto                             
        │   └───leds                         
        ├───passive                          
        │   ├───capacitors                   
        │   ├───magnetics                    
        │   └───resistors                    
        └───transistors                      
```

My System Details:  
- Windows 10 Pro x64
- Version 1709
- OS Build 16299.125

atoav, 2018-02-12:
Also my own compile (from yesterday) exits with: `terminate called after throwing an instance of 'Gio::ResourceError'` in Mysys, this might be of use?

peterus, 2018-02-12:
hi!
this artifact is from my last change: #111 
looks like there are still some problems, which i have to debug.

please use the last stable "release":
https://ci.appveyor.com/project/carrotIndustries/horizon/build/1.0.90/artifacts

atoav, 2018-02-12:
Indeed!
@peterus the 1.0.90 artifact doesn't throw this error

peterus, 2018-02-12:
is there a error in the console? i hate this windows boxes - you get no info from them :)

atoav, 2018-02-12:
I tried running your builds with MYSYS (both the ApVeyor-artifact as well as the one I compiled on mys system) and after crash I got: `terminate called after throwing an instance of 'Gio::ResourceError'`

peterus, 2018-02-12:
ok thanks, i will cross check!

peterus, 2018-02-12:
found the issues! commit will be delivered just in some min.
I hat that there are no errors coming at compile time when the references are not correct.

carrotIndustries, 2018-02-13:
seems like this is fixes

closed 2018-02-13 by carrotIndustries

peterus, 2018-02-13:
just for completeness: finally i could test it, and it is working now! -> fixed