horizon-eda/horizon #113: Package Editor: invisible Pads selectable

atoav, 2018-02-12:
I think this is unwanted behaviour:  
if I switch off the Pad Layers and toggle on `Select only on  work layer` I can still select the (invisible) Pads which can get in the way of doing drag selections:  
[image]

carrotIndustries, 2018-02-13:
The reason why it is that way: Pads don't have a specific layer and the code is kinda stupid in that place right now.
What's your expectation regarding selecting pads in "select only on work layer" mode depending which layer is the work layer?

atoav, 2018-02-13:
I expect selection to work this way:  
- If the `Select only on work layer`-option is **off** I should only be able to select objects on visible layers.
- If the `Select only on work layer`-option is **on** I should only be able to select objects on selected layers.
- The selected layer should be always visible.
- Every selectable object must be on a layer.  

IMHO the pads should also be on layers, it makes no sense for them not to be, everything else makes no sense.

carrotIndustries, 2018-02-13:
> Every selectable object must be on a layer.

That's trivial for things like lines, polygons and tracks, but gets more difficult with composite objects such as pads vias and packages since these span multiple layers. Holes also don't have a layer in the typical sense.

atoav, 2018-02-13:
I can imagine that this can be a non-trivial problem, that would probably demand a fundamental rewrite of a big part of existing code and should be well planned and executed if we decide to go that route.  
I don't think it would become any easier after 0.1 tho..

carrotIndustries, 2018-02-19:
Another idea: how about not selecting non-layer things at all in "work layer only" mode?

atoav, 2018-02-19:
That makes sense. I realized today when fixing the TSSOP packages, that the I cannot toggle the visibility of the (red) top layer for the pads. Is this on purpose?

carrotIndustries, 2018-02-20:
The work layer is always visible to hopefully avoid confusion when cycling through layers.

atoav, 2018-02-20:
The work layer's eye symbol should also represent that (the eye symbol should never be crossed out and grey when it's layer is actually visible).  

**Solution:** Everytime a layer becomes _work layer_, set its visibility to True, switch the eye symbol to opened and block the clicking on the eye symbol to avoid confusion. When the layer is becoming inactive, just leave it visible.

Additionally another checkbox option like `☐ Show only work layer` would also make sense, which would be especially cool to just inspect a packages layers one by one quickly. Then obviously every layer has to become invisible, when it is deselected. It would probably also make sense to register global hotkeys for layer selection (I propose <kbd>PageUp</kbd> / <kbd>PageDown</kbd>) and toggle layer visibility (maybe sth like <kbd>v</kbd>?). 

Only problem here: I'd expect the software to remember the layer visibility I had _before_ activating `☐ Show only work layer`, when I deactivate it again. Imagine it as a sort of "I'm-confused-let-me-check-the-single-layers-for-a-moment"-option that you just use to inspect something quickly.

carrotIndustries, 2018-02-23:
> Another idea: how about not selecting non-layer things at all in "work layer only" mode?
4ef7c950d0cb43a232d89457f685d4b50eaf6e41 implements just that