horizon-eda/horizon #114: [UI] Package-Editor: Visibility Selection symbols

atoav, 2018-02-12:
Change the checkmark icons in the visibility selection of the Package Editor to something more visually clear. At the same time the whole active layer can be ditched for something that is faster to decode than a small dot.  

[image]

These kind of stuff costs time to implement, but the gain in visual clarity helps all users to quickly find their way into the software and helps seasoned users to put their focus on where it is needed.  

If it should come to this I can supply the icons and supply them as SVG/etc.

carrotIndustries, 2018-02-13:
I've had something similar to this in the past, but it had the side effect that clicking on the visibility thing also changed the work layer. Do you think that is is acceptable?

atoav, 2018-02-13:
I think having visibility affect active would be bad, because typically when a user wants to toggle visibility, they want to do that because:
- they want to see (and select) a layer that was previously invisible
- (more often) they want to stay on the layer they are on and make other stuff invisible.

Don't get distracted by the extend of the black bar, I am not that familiar with GTK, but I suspect one could do something like adding the visibility buttons on top (in css this would be called z-index or stacking order) so they block the click to the active button/column.  

From a pure usability standpoint I'd love to see that in combination with dragable lists as [I mentioned in issue #103](https://github.com/carrotIndustries/horizon/pull/103#issuecomment-364750710). Something like they use in Photoshop or Blender:  
[image] [image]  

I know that this kind of drag click thing could take a day to implement, but it could potentially be reused at multiple spots (like the Selection Filter Window etc). Also you wouldn't need `Select All/Select None` Buttons anymore, because click-dragging is just as fast.

greenscreenflicker, 2018-02-13:
@atoav I love that idea, looks amazing from a usability standpoint of view.

jeanthom, 2018-02-13:
What does the "I" column means ? Is that the layer level ?

carrotIndustries, 2018-02-13:
By using a GtkListBox instead of Treeview, we should be able to get around the limitation that clicking changes the work layer. I'll still need to investigate whether the click-and-drag interaction is feasible to implement.

> What does the "I" column means ? Is that the layer level ?
It's the internal layer index. It will disappear before release 0.1

carrotIndustries, 2018-02-19:
af284b67f23951f4504dc1ca61f25fe6ccb8eeed implements some of the suggestions. Comments welcome.

atoav, 2018-02-19:
Great thing @carrotIndustries, checked it out right away in a fresh compile and it really makes things a lot easier. I see you also cleaned up the columns, great job.  

That would only leave the draggable list feature open, but I think that one isn't too urgent and it should be implemented in a more generic way so it could be reused in other places if needed.

carrotIndustries, 2018-02-19:
The "drag to toggle" is kinda difficult to implement since the widget that received the button press event grabs all subsequent events.

atoav, 2018-02-19:
Okay, I understand, this would probably be a whole new custom widget, that somebody would need to sketch, code, test and maintain. From a UI standpoint it still would make sense, but I think there is probably other important stuff too. It is not too urgent tho.

jeanthom, 2018-02-20:
Asked for help on #gtk+ (GIMPNet) about Click and drag implementation in GTK+ 3 :

> you can't do that. If you want this kind of behavior then 1) it will always be ugly 2) you basically need to add a drag gesture to the parent widget and check which one the drag hovers

carrotIndustries, 2018-02-20:
Thanks for asking. Closing this since "drag to change visibility" is unlikely to arrive in the near future.

closed 2018-02-20 by carrotIndustries