horizon-eda/horizon #116: [UI] Package Editor: Courtyard Parameter Voodoo

atoav, 2018-02-19:
The courtyard parameters stuff could be implemented better. Right now the user has to:
1. Draw a courtyard polygon on the right layer
2. Type `courtyard` into its parameter field
3. Click the `Parameters...`-Button to open the `Parameters`-Window
4. Paste some text from a website
5. Get the courtyards polygon lengths via `Enter Datum`
6. Correct it to a syntax the compiler understands
7. Click the `Add Parameter`-Button
8. Change its default value to the thing the website says
9. Click `Apply`

I think a lot of that stuff should actually be done by default if this is something that we _require_.  

If I add a poly rectangle to the top Courtyard layer and give it the parameter class `courtyard` – which it could also get automatically, because what else would you add on that layer – then why do I need to _manually_ get the dimensions of the rectangle with `Enter Datum` to enter it in the Parameters window? On top of that `Enter Datum` seems to return floating point numbers with a comma, while parameters wants them to be with dots (see #115 ) and therefore throws a `Compile error: unhandled token 1,234`. So you change `1,234 mm` to `1.234 mm` which _again_ throws an error because it obviously needs to be written without space between ` 1.234` and `mm`. Then you have to _manually_ add a courtyard expansion parameter (if it is required why not add it automatically), which defaults to `0.500 mm` while the wiki says we should default to `0.250 mm`. And when I click `Apply` there is no real feedback for it beeing applied, other than the courtyard changing.  

I don't know C++ that well, but a bit of Rust and Python, so _I_ found out quite quickly, but other users? Will they remember the (for us logical) order of x and y? Will they notice the space or the comma? Will they find the right website to paste from?  

 I know that it should be possible to automate that by default, even for non-rectangular shapes – people would still able to edit the stuff if they like to. But minimizing the gotchas and stuff people have to remember in order to create a part that fits our rules is crucial. There msut be a better way : )

carrotIndustries, 2018-02-20:
Oops, looks like I forgot to update the wiki back when I added the courtyard program generator...
https://github.com/carrotIndustries/horizon/wiki/Creating-a-Package#courtyard 

74649ca0bcecf0e77905251a6cdb4d49d6cced9a automates creating/setting the parameter.

Automating the creation of the courtyard polygon itself should arrive soon as well.

atoav, 2018-02-20:
That's already a great improvment, I can probably test it tomorrow

carrotIndustries, 2018-02-23:
66741134149810018f37f98233252651a1147a80 not quite a wizard, but in combination with the improved dialog, this should remove most of the voodoo