horizon-eda/horizon #119: [UI] IMP : add a toolbar for selecting tools

jeanthom, 2018-02-21:
Currently most tools are only available through keyboard shortcuts. Having a toolbar with all the tool (draw net, place text, etc.) could improve the learning curve of the schematic and board editor.

carrotIndustries, 2018-02-21:
> Currently most tools are only available through keyboard shortcuts

There's always the spacebar menu providing a searchable list of all tools with their shortcuts right next to them.

TBH, I don't like toolbars at all. After some time they usually end up looking like this: https://i.stack.imgur.com/tPIf4.png

At first sight, many of the icons could be hieroglyphs as well without reducing their meaningfulness, so the long-term user interaction goes like this: The user hovers over all icons to learn from their tooltip what clicking on them will do. After some time, she remembers "clicking on this white-blue colored blob will recalculate all cooper planes".  By that time she could have remembered "press q to recalculate all copper planes" as well.

The spacebar menu itself may be a bit hard to discover indeed, but once the user found out about it, all tools (soon actions) are easily discoverable.

jeanthom, 2018-02-23:
Well, maybe we shouldn't include all the tools available in the toolbar, but at least the core features.

I don't know much about Cadence software (last time I used it was like a year ago, and for a very short period of time). On other EDA software I've used (Eagle, KiCad), the main issue with toolbars is that everything is crammed in (both tools selection & tools properties). That makes the whole toolbar crowded and indigestible to a newbie.

I'd like to hear feedback from other users about shortcuts and spacebar menu.

atoav, 2018-02-24:
I must say I am on the same side as @carrotIndustries on this topic, I would avoid a toolbar with a ton of symbols – I think promoting the spacebar menu and well thought shortcuts makes more sense.

If there really should be a mouse-only way of selecting commands or tools, I could imagine a well organized nested menu like for example `Edit → Copy` and `Draw → Line` etc. But I dont really think that is necessary. I drew ~30 packages with the current version and the Spacebar Menu did it's job quite well for me. Nested menu doesn't necessarily mean dropdown menu, I will come back to this below.

IMHO in UI-Design there is often a tradeoff to be made between "unexperienced user doesn't get threatened" and "experienced user doesn't gets slowed down". I must say that 100% of the software I really _like_ to use in my daily work falls into the latter category and software that falls into the first can become quite annoying if you really use it seriously. The best experiences I had with any UI were at the same times also the ones with the hardest to learn interface (assuming we are talking about higher complexity software and not utility tools that just need to do one job). Having a good interface for pros and beginners alike are not mutally exclusive goals however and having a **consistent** UI is key here.  

Consistency is achieved by implementing similar strategies in multiple parts of the software and rely on conventions where they make sense. For example in Blender you move stuff with the grab command <kbd>g</kbd> – and this is true for every part of the software where stuff can be moved, including timelines and 2D stuff, and if you press <kbd>x</kbd> after pressing <kbd>g</kbd> you will lock the X-Axis, and this again works the same way in all parts of the software. This way knowledge you get in one part of the software transfers easily to other parts. This ability to chain single key commands can get very very powerful if implemented in the right way (every vim user will know) and displaying the available single key commands on bottom is the right way to go.  

For CAD software I kinda like the speed and overview you get from a "Pie Menu" or "Marking Menu", the software that did this the best was Alias StudioTools (now owned by Autodesk). [Check this video](https://www.youtube.com/watch?v=dtH9GdFSQaw), where one of the devs explains why they chose a marking menu instead of a traditional menu. In their production software you would typically press <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>Click</kbd> to get a Marking Menu. And for each mouse-button you'd get a different menu (Selection stuff is on LMB, Edit stuff on the RMB and View stuff on the MMB). Because you can do stuff gesture based and it gets in your muscle memory working this way can get fast as hell.  

I am quite a fan of their approach, but I think cleverly chosen single button shortcuts are just as fast and in combination with the Spacebar Menu I have no worries about people finding stuff : )

niclashoyer, 2019-05-09:
I must say that I love the space bar menu! There needs to be a better introduction, though. I tested LibrePCB before and it really took a lot of time until I understood how I could draw a damn line in horizon.

atoav, 2019-05-09:
I agree, this is still work in progress. I feel like the best we've got at the moment is the "Basic Editor Usage" Section of the [wip branch of the ducumentation](https://horizon-eda.readthedocs.io/en/wip/tools.html).

If you go through the whole chapter it should decently equip you with the basics.

carrotIndustries, 2019-05-12:
> There needs to be a better introduction, though. 

How about another window similar to #250?

carrotIndustries, 2020-05-01:
"It's missing the toolbar" is a phrase I've heard quite a lot form user's so I've had a take at implementing a non-crappy toolbar, see branch https://github.com/horizon-eda/horizon/tree/action-bar (only for the symbol editor so far)
@atoav What are your thoughts?

[image]

atoav, 2020-05-01:
Although I don't really need it, I have nothing against it. The way you did it reminds me very much of the way (you guessed it) Blender solved the same issue:  

[image]  
[image]  

As you can see they put the most common tools for a given context into the tool bar, they grouped similar tools (long-click onto symbols with the triangle in the corner to fold out). Interestingly enough Blender also has a spacebar menu, but they still decided to put a more complete command list into the menu bar on top of the viewfinder.

In my judgment the by far hardest task when it comes to toolbars is to come up with a way to decide what goes in there, what doesn't, how it is organized and which symbols represent it.

I don't have too much spare time right now, so I can't do it, but what would be interesting to have is a simple list of tools for any given context, so we can decide how they could/should be organized and what solution would be the best fit.

In any case showing good tooltips (with shortcuts!) and having good icons is also crucial.
Right now you are hiding the toolbar once a tool is selected – I am not so sure about that to be honest. In Blender you can hide/unhide the toolbar by hitting <kbd>t</kbd>, maybe something similar could also be a option in horizon

carrotIndustries, 2020-05-01:
> The way you did it reminds me very much of the way (you guessed it) Blender solved the same issue

That's no accident as that was my main source of inspiration :)

> Right now you are hiding the toolbar once a tool is selected

That's itentional, since I want to have maximum space for the document if a tool is active and these buttons would be disabled anyhow.

> In my judgment the by far hardest task when it comes to toolbars is to come up with a way to decide what goes in there, what doesn't, how it is organized and which symbols represent it.

My plan was to only put the most relevant tools and actions without obvious shortcuts (such as copy/paste) in that bar. That bar is also supposed to only contain actions that manipulate the document int some interactive way rather than just popping up a dialog.

> In Blender you can hide/unhide the toolbar by hitting

There'll be a setting to turn off the toolbar for users who don't need it.

> to put a more complete command list into the menu bar on top of the viewfinder.

Having thought about this for a bit, I'm not sure if we might need something like this as well. With the action bar, we'll have these ways to trigger actions:

 - spacebar menu (has all of them)
 - hamburger menu (a few ones with global scope)
 - context menu (actions that apply to the selection)
 - action bar (select important actions)

Some useful actions such as the airwire or  selection filter are currently only reachable using the spacebar menu. While this is really convenient if you know what you're looking for, I'm not sure if it makes actions as discoverabie as they should be as it's quite small in size.

Right now, I have in mind to make the action bar toggleable into a larger version of the spacebar menu to provide a more easy to read overview of all available actions.

atoav, 2020-05-02:
> Some useful actions such as the airwire or selection filter are currently only reachable using the spacebar menu. While this is really convenient if you know what you're looking for, I'm not sure if it makes actions as discoverabie as they should be as it's quite small in size.

I think making such cool actions discoverable is a value in itself. My feeling is that the action bar however should only contain tools that allow you to manipulate or create something (we could like Blender give these distinct colors by type). Actions that affect the view itself (e.g. filters, View all, View Bottom, ...) could then maybe have another place on the display again inspired by the way Blender does it:  
[image]  
This has the obvious advantage that once users learned that view related stuff is in the top right corner, they will check that corner if they want to change view-related settings. Additionally a nice "clear all view filters"-Button could be added. This could be useful across all editors. 

Note how blender has round "Hand"- and "Magnifier"-Buttons on the right? You can click and drag those to zoom in or pan around (Blender's tooltip doesn't show users the way to do it with mouse and keyboard tho, which is a pitty)

# What needs to be done (asides from programming)
## Structuring
thinking about how to structure tools for all editors, this includes thinking about additional UI features like a "view bar" as mentioned above or "fold-out" type of buttons, as this impacts which things go where and how things are grouped.

Obviously there are multiple factors affecting how useful it is to place a tool into the action bar here. Things that are obvious (like undo) don't need to be in there. Things that first time users will want to do, but have a hard time discovering are useful. Things that are cool but hard to discover as well.

How does the space-bar menu tie into this? maybe there could be a dedicated button to call it as well? I would never use it, but it might help to onboard users which are new and don't know about the spacebar thing.

## Designing
This topic involves things like the look of icons, the size and positioning of the UI elements, whether colors are used, etc.

If we hide the action bar if a tool is active, it becomes important to make it easily recognizable  that you are within a tool context – otherwise there will be people who will wonder where their tools went etc. Right now this is signified mostly by the context-bar on the bottom, which is visually easy to overlook. Maybe this could be made more prominent (purely by size and contrast), or we could consider additionally adding something like a border around the whole viewport could make this even more obvious to users:

[image]
(Note that I used symbols for LMB/RMB and adjusted font size and spacing a bit)

I can do the icon design if you want and mock some things up : )

atoav, 2020-05-02:
As proposed, I try to create a structure of what things to put where. I almost definitly missed something, feel free to expand.

# All editors (?)
### Action Bar
- <kbd>r</kbd>: Rotate
- <kbd>e</kbd>: Mirror
- <kbd>p</kbd><kbd>t</kbd>: Place Text
### View "Bar"
- <kbd>Home</kbd>/<kbd>`</kbd>  View all
- <kbd>Ctrl</kbd>-<kbd>Home</kbd>/<kbd>Ctrl</kbd>-<kbd>`</kbd>  View selected (Not yet implemented, but useful?)
- <kbd>Ctrl</kbd>-<kbd>I</kbd>: Selection filter
- <kbd>Shift</kbd>-<kbd>LMB</kbd>: Pan
- <kbd>Scroll</kbd>: Zoom

# Symbol Editor
### Action Bar
- <kbd>p</kbd><kbd>T</kbd>: Place Refdes and Value
- <kbd>p</kbd><kbd>p</kbd>: Place Pin
- Draw line commands (fold-out):
    - <kbd>d</kbd><kbd>a</kbd>: Draw arc
    - <kbd>d</kbd><kbd>l</kbd>: Draw line
    - <kbd>d</kbd><kbd>L</kbd>: Draw line rectangle
- Draw polygon commands (fold-out):
    - <kbd>d</kbd><kbd>y</kbd>: Draw polygon
    - <kbd>d</kbd><kbd>Y</kbd>: Draw polygon rectangle
    - <kbd>d</kbd><kbd>c</kbd>: Draw polygon circle

# Schematic editor  
### Action Bar
- <kbd>p</kbd><kbd>p</kbd>: Place part (this can then be removed in the top bar)
- <kbd>p</kbd><kbd>o</kbd>: Place power symbol
- <kbd>p</kbd><kbd>p</kbd>: Place component
- <kbd>b</kbd>: Place Net Label 
- <kbd>a</kbd><kbd>a</kbd>: Annotate 
- Draw line commands (fold-out):
    - <kbd>d</kbd><kbd>a</kbd>: Draw arc
    - <kbd>d</kbd><kbd>l</kbd>: Draw line
    - <kbd>d</kbd><kbd>L</kbd>: Draw line rectangle
- <kbd>Ctrl</kbd>-<kbd>Tab</kbd>: Go to Board

## Board Editor
### Action Bar
- <kbd>x</kbd>: Route
- <kbd>v</kbd>: Place Via
- Draw line commands (fold-out):
    - <kbd>d</kbd><kbd>a</kbd>: Draw arc
    - <kbd>d</kbd><kbd>l</kbd>: Draw line
    - <kbd>d</kbd><kbd>L</kbd>: Draw line rectangle
- Draw polygon commands (fold-out):
    - <kbd>d</kbd><kbd>y</kbd>: Draw polygon
    - <kbd>d</kbd><kbd>Y</kbd>: Draw polygon rectangle
    - <kbd>d</kbd><kbd>c</kbd>: Draw polygon circle
- <kbd>d</kbd><kbd>d</kbd>: Draw dimension
- <kbd>Ctrl</kbd>-<kbd>Tab</kbd>: Go to Schematic
### View "Bar"
- Airwire filter

carrotIndustries, 2020-05-02:
> r: Rotate

I don't think that this should go there as it's a contextual action that's already easily discoverable from the context menu. It's also quite unergonomic to move back and forth between the edge of the viewport and the thing that's being worked on for such basic actions.

> View "Bar"

Perhaps this could go into a hamburger menu right next to the other one?

> pp: Place Pin

I don't think we really need this in the action bar as we already have the list of unplaced pins that should make it obvious how to place pins.

atoav, 2020-05-02:
I am not really against providing multiple ways of invoking a single tool, if there is a benefit to it. however your point on the Rotate and Place Pin tool is spot on, I hadn't really thought about that.

On the topic of the View Bar: 
> Perhaps this could go into a hamburger menu right next to the other one?

Intuitively this would fit much better into the viewport itself, as it affects the viewport while the stuff in the existing hamburger menu has a global scope. I will try to draft something up, so we can see how this could look.

carrotIndustries, 2020-05-02:
> Intuitively this would fit much better into the viewport itself, 

Keep in mind that quite a few corners of the viewport are already occupied by various items that could pop up based on context:

 - top right: search
 - top center: reload pool for symbols and packages
 - top left: hud
 - right: action bar
 - bottom: tool bar

> Note that I used symbols for LMB/RMB and adjusted font size and spacing a bit – in the future the color of this frame could even be used to signify something is Ok, wrong etc

The thing is that the tool bar is quite crowded in some tools that have quite a lot of options such as the draw polygon rectangle tool.

I've thought about adding some sort of panel to the right that replaces the inspector for tools that need actual UI for changing tool settings rather than just a list of shortcuts. For the draw polygon rectangle tool for example, there'd be a dropdown menu to select the style of decoration and a spinbutton to change the size of the decoration.

But let's do one thing after the other.

carrotIndustries, 2020-05-02:
Padstack WIP:
[image]

atoav, 2020-05-02:
Good points, how about moving the search to the top center – if anyone reloads the pool during search, the search window could just be hidden, then we could use the top right for filtering like this:
[image]  

As for the bottom tool bar: I actually quite like it conceptually, it was a very good idea and works extremely well for the whole control concept you used throughout horizon, as it gives users additional context about the tools currently in use.

The Mockup I did was probably a bit to generous with space, but I think even with the `Draw Polygon Rectangle` we could get a way with a bit more generous spacing (this emulates a display with a width of 1440 px):
[image]  
Here I just made the tool name 0.5 pt bigger, used square brackets for keys and added for spaces after each thing to seperate it visually.

As for the padstack wip: the icons look quite good to me. Maybe the style could be a little bit more modern however?  

[image]  
What do you think of something like this?

carrotIndustries, 2020-05-02:
Action buttons are now free-floating and use larger stroke width to better complement other gnome icons:
[image]

> Good points, how about moving the search to the top center

Pretty unusual place for a search overlay. Chromium, gedit and gnome-builder place it in the top-right corner, so that's where I and probably other people expect it to show up.

atoav, 2020-05-02:
> Pretty unusual place for a search overlay. Chromium, gedit and gnome-builder place it in the top-right corner, so that's where I and probably other people expect it to show up.

In Firefox it shows up on the bottom left (as does it in most text editors, including Word). OS X spotlight search is even centered on screen. I don't really think there is that much of a convention to hold up to be honest. As long as something pops up and you can see it it is pretty much okay – especially if it catches the text-focus automatically as horizon does.

Moved to the center it would look like this:  
[image]  

But this is of course more permanent than the search window and would potentially collide with the Pool Reload indication. I don't really have the feeling that you are very fond of the whole thing at all – do you have the feeling it occupies to much screen space?

I think the free floating action buttons look better now. As for the stroke width, I feel that thinner strokes look more elegant (and I don't use Gnome so conforming to it isn't really a priority for me). The icons you made for the padstack looked quite good to me.

Just tell me if you need any help creating icons (and what format you need them in if you do).

carrotIndustries, 2020-05-02:
>  However tastes differ etc.

That's probably the case, to me thin-stroked icons like https://docs.microsoft.com/en-us/windows/uwp/design/style/segoe-ui-symbol-font look like they'd fall over on the lightest gust.

> I don't really have the feeling that you are very fond of the whole thing at all – do you have the feeling it occupies to much screen space?

Why not stick the icons in the headerbar as we have plenty of space available in there?

> Just tell me if you need any help creating icons

To me drawing icons, is kind of a relief from coding, so I think we're fine in that regard. Anyhow, feel free to point out any ideas for improvement.

atoav, 2020-05-02:
> That's probably the case, to me thin-stroked icons like https://docs.microsoft.com/en-us/windows/uwp/design/style/segoe-ui-symbol-font look like they'd fall over on the lightest gust.

Sure, if you feel that way, we better go with something more stable-looking.

> Why not stick the icons in the headerbar as we have plenty of space available in there?

Because it suggests a different kind of context (global vs viewfinder) and quite frankly would look a bit odd. Before putting it there I'd rather not include it at all to be honest.

> To me drawing icons, is kind of a relief from coding, so I think we're fine in that regard. Anyhow, feel free to point out any ideas for improvement.

Understandably so, this is quite similar for me. I will come back at you if something hurts my sense of aesthetics (and it probably will, as expected from a MA of Arts 🥇)

carrotIndustries, 2020-05-03:
> Because it suggests a different kind of context (global vs viewfinder) and quite frankly would look a bit odd. Before putting it there I'd rather not include it at all to be honest.

Perhaps it's because to me at least some of the icons aren't immediately clear on first sight:
 - Funnel: okay, selection filter / menu for airwire filter
 - House: view all
 - Hand: drag??? well, we have drag using the middle mouse button
 - Loupe: zoom??? we have mouse wheel zoom
 - Pointer: select???

Anyhow, we now have the possibility of adding pull-out menus to the action buttons and I did some further icon tweaking since the place shape icons looked to heavy compared to the other ones.
The menu can be activated by:
 - Clicking on the tiny arrow on the bottom right of the button
 - Right-clicking on the button
 - Dragging to the right (loosely inspired by blender)
 
[image]

atoav, 2020-05-03:
> Perhaps it's because to me at least some of the icons aren't immediately clear on first sight:
> - Funnel: okay, selection filter / menu for airwire filter
> - House: view all
> - Hand: drag??? well, we [...]

Because it would be not very useful to mock a menu up with only two icons if we are interested in "how things could potentially look in the future" I made a few things up – sorry if that was unclear.
 
The hand and the magnifier glas are both things that come more or less from Blender directly as I mentioned in [a previous comment](https://github.com/horizon-eda/horizon/issues/119#issuecomment-622890286). These are not _traditional_ buttons: clicking and dragging them zooms/pans the view. This is useful in situations without mousewheel and helps onboarding new users who don't know how to pan yet (e.g. I showed horizon to multiple people, and some of them had issues figuring out that MMB pans the view).

The Pointer symbol was a "invention" as well meant to be a button for a fictional `view selected` action, which I only mentioned very briefly in [the list above](https://github.com/horizon-eda/horizon/issues/119#issuecomment-622946500) so I don't blame you for overlooking it.

With this kind of mockup I usually try to convey a feeling for how the space is used, which means it is usually more useful to "skip ahead" to a future where more tools have been added. So this wasn't meant as a 100% accurate depiction of how things should look but should rather give a feeling of where things could potentially go.

## On the action bar

As for your menu implementation:
- I like the dragging thing
- RMB is also intuitive
- Clicking the corner also works, although it is a bit more obscure

Some nitpicks:
- Many users would probably still try to long-click the thing (but I understand that counting time is not a thing that is always easy to implement)
- It is probably better to list _all_ the available commands (so in this case `Draw polygon` should be displayed as well). This is better because it allows users to change their mind and still decide to draw a polygon instead.
- The expectation would probably be to see a polygon-circle symbol if you select the `Draw polygon circle` action once. So selecting another tool should mean that that tool is now the new "default" in the toolbar. At least this is how it is done in most existing software that uses this kind of pattern (e.g. Blender, Photoshop & the whole Adobe suite).

I know this means more work as well, but what do you think of using icons instead of a list as well? E.g. like this (I did a quick shitty mockup):

[image]

carrotIndustries, 2020-05-03:
Thanks for your explanation on the circular icons, let's save these for later as I think that they're in general unrelated to the action bar and let's focus on that one first.

 > Clicking the corner also works, although it is a bit more obscure

Oh well, I though that this one was the one that's most easily discovered as clicking on a tiny arrow is a fairly standard action for invoking a menu (think dropdown)

> Many users would probably still try to long-click the thing

That's what I tried first, but gtk yelled at me for opening a menu without an event and long-clicking isn't something I've often seen on desktop apps. In fact, the only instance where I've encountered this are the back/forward buttons in firefox.

> The expectation would probably be to see a polygon-circle symbol if you select the Draw polygon circle action once. So selecting another tool should mean that that tool is now the new "default" in the toolbar.

I was unsure about that one, but now that I had a look at other apps like libreoffice I noticed that they also implement it as you described, so that's the way to move forward. 

Now we're stuck with having to come up with an icon for 'draw polygon rectangle' that' different from the 'place rectangular shape' perhaps make the shape icons solid again?

[image]

atoav, 2020-05-03:
Ahh great! Having the names there was a even better idea and removes the need to display tooltips there. If we now add the key sequences right aligned to these fields (similar to the way it is already done in the spacebar menu) this could serve as a escape hatch for users who have to use some commands over and over again: they just see the shortcut and can use that instead:

[image]  

As for the symbols I already have an Idea that would increase the clarity a lot, let me try it for a bit I will post the results soonish.

carrotIndustries, 2020-05-03:
> we now add the key sequences right aligned to these field

Well, that was easy: 
[image]

atoav, 2020-05-03:
> Well, that was easy

Cool stuff. 

## Icons

As for the promised ideas on the icons:
- For the polygon tools it makes sense to highlight everything that makes polygons.. well polygons. To me that are two things: vertecies (as you have to lay them down) and the possibilty to create convex shapes
- For the hole tools it makes sense to add perspective (to highlight these make actual holes in the PCB), this again helps to differenciate them from the rest
- Because the former two are now visually distinct from the place shape tools, these can stay as they are

Another idea is to colorcode the icons to quickly differenciate different categories, but that is a matter of taste, how well defined the categories are and how much legibility is ultimately worth.

| Icon | Colored Icon | What is it |
| --- | --- | --- |
|  [image] | [image] | Draw polygon |
| [image] | [image] | Draw polygon rectangle |
| [image] | [image] | Draw polygon circle |
| [image] | [image] | Place hole |
| [image] | [image] | Place slot hole |
| [image] | [image] | Place circulare shape |
| [image] | [image] | Place rectangular shape |
| [image] | [image] | Place obround shape |

As you can see I tried to stay very much within the style you proposed, but tried to make the shapes more "readable", especially in the ensemble. The colors I proposed there are by no means set, I just chose _something_ to get stuff done.

carrotIndustries, 2020-05-04:
Adding the control points to the polygons really helps to tell them apart from the shapes. Great idea! I'd go with the non-colored icons as the color might clash with the background color shimmering through the button.

How to move forward with this? The icons themselves are 16×16 px SVGs that get scaled up to 32×32px. Everything should be aligned to a 1 px grid (½ px is fine as well) Do you want to provide ready-to use icons or should I have a go at drawing them to fit the icon requirements? Also see the the Adwaita icons for refrence: https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/blob/master/src/symbolic/gnome-stencils.svg You're only supposed to use `#bebebeff` as fill/stroke color since gtk will automatically recolor icons to match the theme. To achieve opacity, set the opacity of the entire object, rather than on the color itself.

I'm a bit torn about the use of perspective in the hole icons as it took me a bit to realize what I'm actually looking at. The GNOME HIG also discourages the use of perspective: https://developer.gnome.org/hig/stable/icon-design.html.en

atoav, 2020-05-04:
> How to move forward with this? 

I could export the icons as svg and add them to the horizon-artworks repo? I have to go to work today, so I could only doe it later this day or tomorrow.

> You're only supposed to use #bebebeff as fill/stroke color since gtk will automatically recolor icons to match the theme. To achieve opacity, set the opacity of the entire object, rather than on the color itself.

I think this can be done – however I am unsure whether coloring based on the OS-theme is the smartest thing to do here, since the background it sits on is decided by the color theme of the respective editor etc.

> I'm a bit torn about the use of perspective in the hole icons as it took me a bit to realize what I'm actually looking at. The GNOME HIG also discourages the use of perspective: https://developer.gnome.org/hig/stable/icon-design.html.en

Discouraging perspective from icons makes sense in a OS-context. This is okay unless the very nature of what the icon is meant to communicate is more than two-dimensional (e.g. a hole in a PCB, a bevel tool in Blender etc). This is why I went for the least amount of perspective necessary that gets the job done (ortographic projection, no unnecessary angeling). If I had to decide between clearly communicating what a tool is while breaking a guideline or honoring a guideline and ending up with a icon that makes people guess, you probably know where my decision would fall : )

Btw. for reference, the thing that I predicted to happen, happened (user beeing unable to figure out zoom and pan and turning away): <https://news.ycombinator.com/item?id=23064116>

atoav, 2020-05-04:
Okay added them here: https://github.com/horizon-eda/horizon-artwork/tree/master/icons

I didn't do a proper pixel-grid-alignment yet, but maybe it is good for initial testing. Only `#bebebeff` and object opacity should be used (unless Illustrator export mangled it).

carrotIndustries, 2020-05-05:
> Btw. for reference, the thing that I predicted to happen, happened (user beeing unable to figure out zoom and pan and turning away)

Perhaps I that user and I used a vastly different set of applications before. Every 2D-related application I've used so far (PDF viewers, Inkscape, Gimp, Librecad, kicad, eagle, many more) use middle click to pan and (ctrl) scroll to zoom. To me at least these interactions are universally understood on the same level as Ctrl+s to save.

> unless Illustrator export mangled it

Inkscape got a bit confused about the stroke being unset rather than none, but apart from this, I was able to integrate a few of them into the UI:

[image]

I don't know what you think, but the depth effect on the place hole button isn't immediately obvious. Perhaps that'll get better when the outline is aligned to pixels. You can also make use of the full width, Gtk will take care of padding.

atoav, 2020-05-06:
> I don't know what you think, but the depth effect on the place hole button isn't immediately obvious. Perhaps that'll get better when the outline is aligned to pixels. You can also make use of the full width, Gtk will take care of padding.

I think I have to increase the size of the hole relative to the rest, to make it readable at that small scale, aligning to pixels will also help. I am not sure If I can get that tdone today, but it is on my list.

carrotIndustries, 2020-05-07:
Package editor gained an action bar:

[image]

Should we add a generate silkscreen/courtyard/package to the action bar as well?

atoav, 2020-05-08:
> Package editor gained an action bar

Great!

> Should we add a generate silkscreen/courtyard/package to the action bar as well?

I think that would be neat. These icons pose more of a challenge tho, any ideas? I still have to work today, so maybe I can get to it at the weekend.

fruchti, 2020-05-08:
I really like the direction this is taking! The Photoshope-esque collapsed menus should be pretty intuitive while space-saving.

Personally, I haven't really missed a toolbar but it can be a great discoverability aid. Therefore I think it would be a good idea to include more specialised and less frequently used, but very powerful functions like generate silkscreen or copy placement, too. New users would be missing out if they haven't come across them in the spacebar menu by pure chance and more advanced users might still use the toolbar for this because you don't need these commands frequently enough to assign a key binding to them. I also typed a synonymous name into the space bar menu before remembering the correct one a few times, so a tool bar button for a rarely used function would help me out there, too.

> I don't know what you think, but the depth effect on the place hole button isn't immediately obvious.

FWIW, I also immediately visualised the hole icons as having perspective, although more in a CAD drawing style, with dashed lines for hidden edges and such. These would probably have been overly detailed for the context, though.

> > Should we add a generate silkscreen/courtyard/package to the action bar as well?
> 
> I think that would be neat. These icons pose more of a challenge tho, any ideas?

Maybe with a few shaded pads and the generated polygon/lines highlighted? Rough sketch:
[image]

carrotIndustries, 2020-05-08:
> These icons pose more of a challenge tho, any ideas?

I've had the idea to put then in a menu invoked by a magic wand button: 
[image]

The individual generators  then wouldn't necessarily need an icon as the wand icon will stay there. This differs from the other buttons, but seems reasonable as people won't need one of these tools repeatedly.

fruchti, 2020-05-08:
This sounds good as well. The wand icon could be a catch-all for all ‘do things automatically’ functions, so less frequently used buttons wouldn't take up as much space.

carrotIndustries, 2020-05-08:
>  I  also typed a synonymous name into the space bar menu before remembering the correct one a few times

How about adding some hidden keywords to action to alleviate this? The same way the windows start menu shows "add/remove programs" when searching for "uninstall".

carrotIndustries, 2020-05-08:
> This sounds good as well. The wand icon could be a catch-all for all ‘do things automatically’ functions, so less frequently used buttons wouldn't take up as much space.

That's exactly what I had in mind: 
[image]

fruchti, 2020-05-09:
> How about adding some hidden keywords to action to alleviate this?

That's a really good idea. Adding terms which users of other systems are used to should really help discoverability. A few alternatives for the board editor which come to my mind:

* Place hole
  * Drill
  * Mounting hole
* Update all planes
  * Fill planes
* Export STEP
  * Export 3D model
* Rules
  * DRC rules
* Run checks
  * Run DRC
* Apply rules
  * Update from DRC rules
* Place package
  * Place footprint
* Place via
  * Switch layer
* View all
  * Zoom to fit
* Draw dimension
  * Measure
* Draw track
  * Route
* Flip view
  * Mirror view
  * Show from underside

carrotIndustries, 2020-05-09:
I had a go at making the spacebar menu available from the action buttons and can't say I'm convinced that this is a good idea: 
[image]

Perhaps an overlay "lost in space? press space to see ...." to inform users about the spacebar menu would be better option? The only advantage the above implementation offers that the list shows more items at once with less scrolling.

carrotIndustries, 2020-05-12:
> The only advantage the above implementation offers that the list shows more items at once with less scrolling.

Mitigated by 1240fb8b81549176888e25af8783c0148d0ca0fb

The schematic editor now also has an action bar:
[image]

carrotIndustries, 2020-05-12:
Board editor:
[image]

Apart from the setting to turn of the action bar, seems like we're good to go.

carrotIndustries, 2020-05-13:
Preferences:

[image]

If no one objects I'll merge this onto master.

atoav, 2020-05-13:
I am fine with this. I believe the icons could (probably) still be refined, but who cares as long as it works they can still be refined later etc.

However I have one suggestion – what do you think about changing the "Keep primary action" thing to "Remember action changes" and "Stay with last used rather than switching back to primary action" (so basically inverting the toggle/meaning). This would be more clear somehow.

closed 2020-05-14 by carrotIndustries

carrotIndustries, 2020-05-14:
Thanks for the suggestion, this is how I implemented it:
[image]

Feel free to open new issues for the remaining points from the discussion such as icon improvement.

commit 136cf038 (2020-05-14): add action bar

also make render_icons.rb work with inkscape 1.0

closes #119