horizon-eda/horizon #120: [Proposal] PCB Modules Blueprints (Factorio Style)

greenscreenflicker, 2018-02-23:
For the automation of layout I'd propose a new component: Modules. 
Let's say you have LDO: It always requires an input capacitor and output cacitor. It would be nice to automate that. You should create a module, lets say LDO, that combines that 3 components: Only the module then has to be wired.

The game factorio does a realy nice job in this. Could you think of implementing this for PCB design?

carrotIndustries, 2018-02-23:
For now' that's out of scope. In the far future, hierarchical designs may provide something similar.

raphaelcasimir, 2020-08-17:
I am working at a company designing their product's PCBs, spoken with electronics developpers and they really like the idea. I have been searching for this feature for ages, in particular something that propagates from schematics to board. You often have the same module with the same PCB layout across multiple projects, re-routing them is really a loss of time.
Having modules would be a joy in horizon.
Thanks a lot for your work, it is of a rare quality.

bob-u, 2020-08-17:
I have not tried It but this feature might be address part of your needs:

https://horizon-eda.readthedocs.io/en/latest/copy-layout-placement.html

carrotIndustries, 2020-08-17:
> I have not tried it, however this feature might address part of your needs:

Unfortunately, this tool doesn't work across projects. However, if there's a real need for it, it could be extended to work across projects. The workflow could then look like this:

 - Layout the block in one project, make sure that each component is assigned a unique tag and they're all in the same group
 - Copy the schematic to the new project, groups&tags will be preserved (group/tag names is still todo)
 - Place all packages of the block on the new board (position doesn't matter)
 - Select all packages of the block in the source board, and invoke the "copy placement" tool
 - Invoke the "paste placement" tool in the destination board
 - Select tracks in the source board, invoke the "copy tracks" too
 - Invoke the "paste tracks" too in the destination board

Perhaps it could also make sense to keep the existing copy placement/track tools as is and have one new "Copy placement & tracks" tool.

bob-u, 2020-08-23:
I could use this function myself when designing controller boards with multiple drivers.

@raphaelcasimir Creating a project with just components and PCB traces you'd like to replicate would probably bring you very close to what you are looking for.

@carrotIndustries Could this function be scripted through python module to lessen number of steps?

`Place all packages of the block on the new board (position doesn't matter)`

It would be nice to be able to place group of components in one go, components selected by selecting symbols on the schematic.

`Select all packages of the block in the source board, and invoke the "copy placement" tool`

Again, selecting packages by selecting symbols on the schematic would be easier and less error prone, IMO.

`Select tracks in the source board, invoke the "copy tracks"`

Would names of the nets copy as well?

raphaelcasimir, 2020-08-25:
> Creating a project with just components and PCB traces
That's what I am trying to avoid by suggested an integration within the pool.
One major difference between horizon and KiCad is the fact that you are encouraged to contribute to the pool and that your contributions are still validated before getting integrated.