horizon-eda/horizon #121: Flatpak support

Murmele, 2018-02-23:
I'm working on creating a [flatpak](https://flatpak.org/) package. For this I would ask if the pool manager should be in the package or should it be manually installed?

greenscreenflicker, 2018-02-23:
You require the pool manager to create new packages...

carrotIndustries, 2018-02-23:
Don't forget to include the horizon-imp binary as well since that's the actual editor called from the pool and project manager.

Murmele, 2018-02-24:
Thank you for you hints, hope to get finish the flatpak package :)

Murmele, 2018-03-23:
Here you can find the Flatpak manifest for horizon.
https://github.com/Murmele/horizon/tree/flatpak
First the branch of peterus "switch_to_cmake" must be merged

carrotIndustries, 2020-10-08:
It's on flatpak https://flathub.org/apps/details/org.horizon_eda.HorizonEDA

closed 2020-10-08 by carrotIndustries