horizon-eda/horizon #122: Debian package

wookey, 2018-02-26:
I have done an initial debian packaging. It is available here:
http://wookware.org/software/repo/
use 
deb [ trusted=yes ]  http://wookware.org/software/repo unstable main
in your apt sources to install it. (apt update; apt install horizon-eda)

Feedback welcome (it could easily be quite broken...)
Outstanding issues before it can be uploaded:
 * man pages needed
 * system dxflib should be used

wotwot, 2018-04-25:
hey thanks for the effort. i just tried to build the current master as well as the debian source package on a fresh stretch system but it fails with some [errors](https://pad.hfbk.net/horizon_on_stretch) that exceed my level of understanding c++. it would be awesome if there were .debs for stretch too.

carrotIndustries, 2018-04-25:
7e3139bef2411799dbbf750c19cf5afb2b7f9b6d fixes this issue. Seems like building c++14 was't a solved problem when debian stretch was released... Can someone enlighten me why this commit fixed things? To me this just looks like the compiler not being as smart as I'd like it to be.

wookey, 2018-04-25:
On 2018-04-25 20:01 +0000, wotwot wrote:
> hey thanks for the effort. i just tried to build the current master as well as
> the debian source package on a fresh stretch system but it fails with some
> errors that exceed my level of understanding c++. it would be awesome if there
> were .debs for stretch too.

Stretch does not have new-enough packages for Horizon to build as you have discovered.

I agree that it would be very handy indeed (then I wouldn't have to
run horizon in a chroot/VM).

My C++ isn't up to working out how hard it would be to make it build
on some earlier version either. I really don't know if the changes
needed are major or minor.

The errors in the attached log show what goes wrong in a (-j1 for clarity) build.

(error of the form " cannot call member function 'void horizon::ImpBase::<foo>' without object <foo>, which is something to do with instantiating a variable before using it.

I don't understand why this works in unstable, but not stable.
Hopefully upstream does. I presume it's something to do with versions.

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

g++ -c -Isrc -I3rd_party -g -O2 -fdebug-prefix-map=/home/wookey/packages/horizon/debian/horizon-eda-0.20180331=. -fstack-protector-strong -Wformat -Werror=format-security -g3 -Wdate-time -D_FORTIFY_SOURCE=2 -D_USE_MATH_DEFINES -fdata-sections -ffunction-sections -pthread -I/usr/include/uuid -I/usr/include/gtkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib/x86_64-linux-gnu/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib/x86_64-linux-gnu/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib/x86_64-linux-gnu/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/x86_64-linux-gnu -I/usr/include/dxflib -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/imp/imp.cpp -o src/imp/imp.o
src/imp/imp.cpp: In instantiation of 'horizon::ImpBase::run(int, char**)::<lambda(auto:2)> [with auto:2 = horizon::CanvasGL::SelectionQualifier]':
/usr/include/c++/6/type_traits:2481:26:   required by substitution of 'template<class _Fn, class ... _Args> static std::__result_of_success<decltype (declval<_Fn>()((declval<_Args>)()...)), std::__invoke_other> std::__result_of_other_impl::_S_test(int) [with _Fn = horizon::ImpBase::run(int, char**)::<lambda(auto:2)>&; _Args = {horizon::CanvasGL::SelectionQualifier}]'
/usr/include/c++/6/type_traits:2492:55:   required from 'struct std::__result_of_impl<false, false, horizon::ImpBase::run(int, char**)::<lambda(auto:2)>&, horizon::CanvasGL::SelectionQualifier>'
/usr/include/c++/6/type_traits:2496:12:   required from 'class std::result_of<horizon::ImpBase::run(int, char**)::<lambda(auto:2)>&(horizon::CanvasGL::SelectionQualifier)>'
src/imp/imp.cpp:229:95:   recursively required by substitution of 'template<class T> void horizon::bind_widget(std::map<T, Gtk::RadioButton*>&, T&, std::function<void(T)>) [with T = horizon::CanvasGL::SelectionQualifier]'
src/imp/imp.cpp:229:95:   required from here
src/imp/imp.cpp:229:90: error: cannot call member function 'void horizon::ImpBase::update_selection_label()' without object
                                                   [this](auto v) { update_selection_label(); });
                                                                    ~~~~~~~~~~~~~~~~~~~~~~^~
src/imp/imp.cpp: In instantiation of 'horizon::ImpBase::run(int, char**)::<lambda(auto:3)> [with auto:3 = horizon::CanvasGL::SelectionTool]':
/usr/include/c++/6/type_traits:2481:26:   required by substitution of 'template<class _Fn, class ... _Args> static std::__result_of_success<decltype (declval<_Fn>()((declval<_Args>)()...)), std::__invoke_other> std::__result_of_other_impl::_S_test(int) [with _Fn = horizon::ImpBase::run(int, char**)::<lambda(auto:3)>&; _Args = {horizon::CanvasGL::SelectionTool}]'
/usr/include/c++/6/type_traits:2492:55:   required from 'struct std::__result_of_impl<false, false, horizon::ImpBase::run(int, char**)::<lambda(auto:3)>&, horizon::CanvasGL::SelectionTool>'
/usr/include/c++/6/type_traits:2496:12:   required from 'class std::result_of<horizon::ImpBase::run(int, char**)::<lambda(auto:3)>&(horizon::CanvasGL::SelectionTool)>'
src/imp/imp.cpp:255:47:   recursively required by substitution of 'template<class T> void horizon::bind_widget(std::map<T, Gtk::RadioButton*>&, T&, std::function<void(T)>) [with T = horizon::CanvasGL::SelectionTool]'
src/imp/imp.cpp:255:47:   required from here
src/imp/imp.cpp:240:72: error: cannot call member function 'void horizon::ImpBase::update_selection_label()' without object
[... cut ...]
             show_preferences_window();
             ~~~~~~~~~~~~~~~~~~~~~~~^~
src/imp/imp.cpp:381:40: error: cannot call member function 'void horizon::ImpBase::show_preferences_window()' without object
                 show_preferences_window();
                 ~~~~~~~~~~~~~~~~~~~~~~~^~
Makefile:502: recipe for target 'src/imp/imp.o' failed
make[2]: *** [src/imp/imp.o] Error 1
make[2]: Leaving directory '/home/wookey/packages/horizon/debian/horizon-eda-0.20180331'
dh_auto_build: make -j1 returned exit code 2

wookey, 2018-04-26:
On 2018-04-25 16:15 -0700, Lukas K. wrote:
> 7e3139b fixes this issue. Seems like building c++14 was't a solved problem when
> debian stretch was released...

Aha, cheers for that. It does indeed work. Very nice indeed to have a
version that builds on stable.

>  Can someone enlighten me why this commit fixed
> things? To me this just looks like the compiler not being as smart as I'd like
> it to be.

No idea. I've never understood c++

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

atoav, 2019-05-28:
@carrotIndustries What will our policy be regarding packaging with the release? I suspect you already had something in mind?

I'd really love to have have up to date Debian packages available, without having to worry whether I will be able to get the current version or something extremely outdated.

A good solution could be to make the packaging/testing part of the CI/CD (just like building for Windows is with AppVeyor). The resulting packages could then be deployed using our own package repository/ppa

To me this sounds like a goal for the actual [1.0 release](https://github.com/horizon-eda/horizon/issues/239), unless there are really hairy gotchas that can't be resolved. It fits with our goal of improving the out-of-the-box experience, because after all the first experience every user will have with the software is how to get it running. For Windows it is already not that hard, even without installer, so I think making it easier for Linux users is even more important.

Could you see this happening?

I see @phlmn is trying to [make it work with OSX](https://github.com/horizon-eda/horizon/issues/271) which would also be a great thing in a way to become truly multi-plattform.

wotwot, 2019-05-28:
> A good solution could be to make the packaging/testing part of the CI/CD (just like building for Windows is with AppVeyor). The resulting packages could then be deployed using our own package [...]

thank you for that suggestion, that would indeed be a very good thing to be able to just apt install horizon to get a current version installed. (no problem if it isn't doable for stretch, buster is fine)

carrotIndustries, 2019-05-28:
The CI runs on debian stretch with backports enabled, so that's not an issue.

How do you envision packing horzion for debian? Add it to debian's repo or maintain a separate repo?
In the first case, what's the deal with new releases in between debian releases? Can we update them or are they frozen once a given debian release has been made? (I feel as if I asked this elsewhere, but can't remember where exactly) I really don't want bug reports against a 1 year old release.

wookey, 2019-05-28:
On 2019-05-28 14:50 -0700, Lukas K. wrote:
> The CI runs on debian stretch with backports enabled, so that's not an issue.
> 
> How do you envision packing horzion for debian? Add it to debian's repo or
> maintain a separate repo?

Horizon is already in Debian's repo, and has been for about a year. 

> In the first case, what's the deal with new releases in between debian
> releases? Can we update them or are they frozen once a given debian release has
> been made? (I feel as if I asked this elsewhere, but can't remember where
> exactly) I really don't want bug reports against a 1 year old release.

A version that is released in debian has a lifetime of about the 2
years of that debian release.

But we also have the opportunity to upload 'backports' newer versions
for that stable debian release (typically matching the versions in
testing/unstable). Nothing forces users to enable backports, but for a
fast-moving package like this we should recommend that they do. So yes
we can update the available version of horizon-eda for debian stable,
as well as for testing and unstable (which is typically whatever has
been uploaded).

It would be helpful to me as packager if you would at least nominate
'good' versions to upload (as opposed to me picking a random day and
uploading that). Ideally you would actually do releases to indicate
this status.

The version of horizon that will get released with the forthcoming
debian stable is from 2018-11-08. It's too late to change that now
(there is an approximately 6-month freeze process), but we can upload
fixed/patched versions if there are actual bugs or security problems. 

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

carrotIndustries, 2019-05-28:
>  Ideally you would actually do releases to indicate this status.

That's the plan at least. Documentation still needs some work.

> if there are actual bugs

There are for sure.

- https://github.com/horizon-eda/horizon/commit/dfa9fad3d884b31c1a987e3d99dd3cf84acdd9a6 not having this and/or https://github.com/horizon-eda/horizon/commit/db12787592138cde53a9ab3c567da0a93d20457d prevents the pool download from working
 - https://github.com/horizon-eda/horizon/commit/a2543c44e8341b4b593ce247b3dc551410ecb306 fixes ratsnest calculation
 - many more

Do we really want to go through every commit since then and figure out if it's a bugfix or a new feature?

wookey, 2019-05-29:
On 2019-05-28 16:38 -0700, Lukas K. wrote:
>     Ideally you would actually do releases to indicate this status.
> 
> That's the plan at least. Documentation still needs some work.

OK. cool.
 
>     if there are actual bugs
> 
> There are for sure.
> 
>   • dfa9fad not having this and/or db12787 prevents the pool download from
>     working
>   • a2543c4 fixes ratsnest calculation
>   • many more
> 
> Do we really want to go through every commit since then and figure out if it's
> a bugfix or a new feature?

Not much. Given the early development nature it's probably best to
treat the buster release as a technology preview, do backports and
recomend that people use those.

On the other hand there will be users of that version for quite some
time and they may well report bugs, in which case it may well make
sense to apply and upload bug fixes if it's not too hard. I don't mind
doing that. We don't have to review every commit between then now,
just known showstoppers and reported issues (no-one has reported any yet).
People are installing it - there are 35 reported installs so far: 
https://qa.debian.org/popcon.php?package=horizon-eda

If we really think that version in buster will do more harm than good
then we could file an RC (release critical) bug and it'll get kicked
out of the stable release (buster), and just continue to exist in
unstable.

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

atoav, 2019-11-11:
Short question on the issue of deb packages: I think it would be amazing if we could somehow add this to the CI that is already happening for the Windows builds – would this be feasible?

wookey, 2019-11-11:
On 2019-11-11 14:15 -0800, David Huss wrote:
> Short question on the issue of deb packages: I think it would be amazing if we
> could somehow add this to the CI that is already happening for the Windows
> builds – would this be feasible?

I've not been following this at the horizon end (and was thinking an
update to Debian was overdue), but if I put the debian packaging onto
salsa.debian.org, then there is pretty much a button to tick to run
CI, and I should do that.

It depends if what you want to check it 'does the last debian
upload still build/test OK', or 'does the current packaging apply
correctly to the current upstream and build'? Salsa will do the
former. Possibly what you are currently using for Windows could/would do the
latter?

But yes, I'm very happy for more CI. 

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

atoav, 2019-11-11:
>  if I put the debian packaging onto salsa.debian.org, then there is pretty much a button to tick to run CI, and I should do that.

Sounds great. I am not in charge of the CI, @carrotIndustries is. AFAIK Appveyor is used for the Windows builds: [See here](https://ci.appveyor.com/project/carrotIndustries/horizon/history). My experience with deb packages is limited, so I am not of great help here.

carrotIndustries, 2019-11-15:
We have some 0.9 test releases that could be used to package something with a proper version number and get everything setup for 1.0.

Having some CI to make sure packaging isn't broken would be great. Perhaps we could also use this to generate packages ourself that people could install if they'd like to use a newer version than what's packaged in debian.

atoav, 2019-11-15:
> Perhaps we could also use this to generate packages ourself that people could install if they'd like to use a newer version than what's packaged in debian.

That was my idea as well. Official packages often lag behind even official releases quite a lot. While this is useful for unixoid system tools with dependencies between them, it can be quite annoying for GUI/Userspace applications at times. Even with big projects like Blender I had issues at times to get the latest release at times, without adding an extra package source.

carrotIndustries, 2019-11-15:
That could also be addressed by a flatpak package. It's just that no one has created one yet.

wookey, 2019-11-16:
On 2019-11-15 14:39 -0800, David Huss wrote:
>     Perhaps we could also use this to generate packages ourself that people
>     could install if they'd like to use a newer version than what's packaged in
>     debian.
> 
> That was my idea as well. Official packages often lag behind even official
> releases quite a lot. 

This is inevitable with distros. Arranging to have a 'good' version in
stable releases is worth putting a bit of effort into.

I've just updated the patches for v0.9.2, and Uwe Steinmann has
volunteered to take over debian package maintainership as he's more
active in this project. He will hopefully do a new upload very soon.

Whilst I'm here I should mention that I messed up the
version-numbering in my early uploads and used 0.<date> instead of
0~<date>. This means we can't just upload '0.9.2' because 0.9.x is a
lower version than 0.20181108 (but would have been a higher version
than 0~20181108). I hang my head in shame, because I am well aware of
this issue with early-stage, date-based versioning in debian, but
somehow used the wrong separator.

Anyway, if you are within a few months of releasing 1.0 (or can be
persuaded to move on to 1.0 sooner in exchange for 'prettier' version
numbers in Debian) then we might as well stick with the date-based
versioning a bit longer, then switch to 1.0 in due course as if I was
actually competent :-) If it's going to be longer than that then we'll
just use the debian 'epoch' mechanism designed to deal with exactly
this sort of thing (so the debian version would be 1:0.9.2).

I've left this choice up to Uwe, but input from you guys on the
schedule and whether you care about this is helpful in deciding.

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

pwnorbitals, 2020-11-13:
Do we have any progress on this issue ?

carrotIndustries, 2020-11-13:
Thanks for the nudge. I'm closing this since we have somewhat out-of-date https://packages.debian.org/sid/horizon-eda packages in the debian repo and upstream-provided up-to-date ones: https://horizon-eda.readthedocs.io/en/latest/installation.html#debian-ubuntu

closed 2020-11-13 by carrotIndustries