horizon-eda/horizon #123: Rename/Naming Action

atoav, 2018-03-04:
When editing a lot of packages I found the (pad) naming field to come in my way more often than not. This has also the reason that the field looses focus when the mouse moves away from it, which is sort of unexpected behaviour.  

I would suggest a broader solution however: a Action that allows for quick editing of a certain default field that is related to Name or Reference Designator. This would of course be something different in every different context (Change Text of a Text Field etc).

carrotIndustries, 2018-03-08:
We already have a tool that does something similar: Enter datum (press enter to start this tool) cbd889bcefebaa3087f381723e42012b3fe67a14 adds editing text and now edits pad names instead of pad position

atoav, 2018-03-08:
Ah very cool, I didn't know that it would change it's functionality with context

closed 2018-03-08 by atoav